ARMmbed/mbed-os-example-cellular #14: Compilation in online compiler fails

sarahmarshy, 2017-06-08:
Fails with this error:
```
Error: Attempt to override undefined parameter 'lwip.ethernet-enabled' in 'application[*]'
```
Removing this line from mbed_app.json - 
`"lwip.ethernet-enabled": false,` 

Results in this error:

```
Error: Attempt to override undefined parameter 'lwip.ppp-enabled' in 'application[*]'
```

You can continue like this until there is no error from mbed_app.json but actual compilation errors, presumably because these features are not turned on/off.

hasnainvirk, 2017-06-09:
Can you provide me with the command you are using for compilation ?

hasnainvirk, 2017-06-09:
I can't reproduce your issue locally (with mbed-cli) ? I don't know how online IDE is wired up. Can you try moving this line:
"target.features_add": ["LWIP", "COMMON_PAL"], 
before any lwip overrides ?

hasnainvirk, 2017-06-09:
Okay @sarahmarshy  I got to the bottom of this. Online compiler uses latest release, i.e., mbed-os-5.4.7 tag.
Cellular APIs will be released in mbed-os-5.5.0. Closing this , as it is irrelevant.

closed 2017-06-09 by hasnainvirk