ARMmbed/mbed-os-example-cellular #18: IAR compiled binary fails to run on Dragonfly

sarahmarshy, 2017-06-09:
I am seeing no output when executing an IAR compiled binary. Though, it will run if exported to IAR and flashed through the IDE. The exported project will be compiled with debugging flags, so there might be an issue there. 

For reference, this is my iar version:
```shell
C:\Repos\mbed-os-example-cellular>iccarm --version
IAR ANSI C/C++ Compiler V7.80.1.11864/W32 for ARM
```

hasnainvirk, 2017-06-09:
@sarahmarshy  There is an issue with MultiTech UDK2.0. Sometimes the binary did not get flashed properly and the software gets stuck in bootloader. Power cycle the device. Plugging-off the power altogather. Reset button wouldn't work at that stage.

sarahmarshy, 2017-06-09:
@hasnainvirk I have tried that multiple times. I am immediately able to flash and run an ARM compiled binary with the board in the same state. I have also seen the behavior you are talking about, usually it is accompanied by a 'fail.txt' file in the drive as well as the binary sitting in the drive without having been 'consumed', as it were.

sarahmarshy, 2017-06-09:
This is happening reliably, even after power cycles.

hasnainvirk, 2017-06-12:
@sarahmarshy Can you please open a Jira Ticket for that ?
I don't have traction for that on Jira. This should be looked into. Thank you for catching it, please elaborate your test setup as well in Jira ticket for reproduciblity.

sarahmarshy, 2017-06-14:
@hasnainvirk Where am I supposed to create a Jira ticket?

hasnainvirk, 2017-06-15:
@sarahmarshy Ashok opened a ticket here https://jira.arm.com/browse/ONME-3109

ashok-rao, 2017-06-15:
Slight correction @hasnainvirk , that ticket was opened by @sarahmarshy only.. :-)

hasnainvirk, 2017-06-15:
@sarahmarshy @ashok-rao I used mbed-os-5.5.0-rc2-oob branch. And it all worked fine. Can you please elaborate what exactly did you see ? Can you still reproduce it with the above mentioned branch ?

hasnainvirk, 2017-06-15:
My steps:
i) Set Path for IAR compile in mbed_settings.py
ii) Checkout mbed-os-5.5.0-rc2-oob branch
iii) mbed compile -m MTS_DRAGONFLY_F411RE -t IAR
iv) Copy the binary 
v) Run ..

Trying now the export ..

hasnainvirk, 2017-06-15:
@ashok-rao @sarahmarshy With a debug build I will get an assertion:
```
mbed-os-example-cellular, Connecting...
Mutex 20019cc0 error -6: Not allowed in ISR context
```
This is a known issue, documented in release notes.  Fix is available here, https://github.com/ARMmbed/mbed-os/pull/4538

hasnainvirk, 2017-08-18:
@sarahmarshy @ashok-rao  What is the status of this after #4538 ?
If the issue is resolved, please close this issue ?

MarceloSalazar, 2017-09-23:
@sarahmarshy could you please confirm fix with mbed OS 5.6 rc2

MarceloSalazar, 2017-09-23:
Please disregard. It's confirmed to work in 5.6

closed 2017-09-23 by MarceloSalazar