ARMmbed/mbed-os-example-cellular #31: Cellular app not working on C030

MarceloSalazar, 2017-09-15:
Application gets stuck:
```
mbed-os-example-cellular, Connecting...
```

Tools:
- mbed CLI: 1.2.0
- GCC_ARM: 6.3.1 2
- App/branch: https://github.com/ARMmbed/mbed-os-example-cellular/tree/mbed-os-5.6-oob

MarceloSalazar, 2017-09-15:
cc @adbridge @hasnainvirk

hasnainvirk, 2017-09-15:
@adbridge We don't have C030 in Oulu. You need to assign it to someone in Cambridge.

hasnainvirk, 2017-09-15:
@MarceloSalazar Call me on skype and we can debug it together, I don't have this hardware  in Oulu

adbridge, 2017-09-15:
@hasnainvirk Marcelo is away now, I will see if anyone else can help

hasnainvirk, 2017-09-15:
@adbridge @MarceloSalazar This may fix the issue https://github.com/ARMmbed/mbed-os/pull/5116

MarceloSalazar, 2017-09-18:
Just to confirm, the example app does work, but it takes a long time to show a response in the serial port.

IMO, this is a 'minor' issue/enhancement. We should display the progress (single counter of re-tries?), but I don't think [trace](https://github.com/ARMmbed/mbed-os-example-cellular/blob/master/mbed_app.json#L6) is appropriate, as it's too verbose and not clear whether the communication is actually working.

hasnainvirk, 2017-09-19:
If the network joining is taking too long, it means heavy shadowing or low power operation. 
How do you power up your board ? USB only ? Or both power adaptor and USB ? Power adaptor is necessary for optimal performance. The problem is the connection retries are handled by the driver, not by the application. So only way to visualise those retries is to enable mbed-trace, which significantly increases the RAM footprint. Please note that mbed-trace and modem-trace are two different things.

MarceloSalazar, 2017-09-19:
> How do you power up your board ? USB only ? Or both power adaptor and USB ? Power adaptor is necessary for optimal performance.

It was indeed USB only, as usual for mbed boards. If there is a strong requirement to power it up using external adapter, then we should add a note to the example docs.

> The problem is the connection retries are handled by the driver, not by the application.

Maybe we could have a background task informing the user that the app is still running?

AriParkkila, 2018-06-04:
@MarceloSalazar I think we can close this issue? Application has now a thread to print dots to indicate an ongoing progress.

MarceloSalazar, 2018-06-04:
Yes, that's fine. Thanks!

closed 2018-06-04 by MarceloSalazar