ARMmbed/mbed-os-example-cellular #40: C030-U201 board doesn't receive always a response from u-blox echo server

gggekov, 2017-12-06:
## H1 Description
The C030-U201 sometimes wouldn't get a reply from the u-blox echo server. Below is an example from the serial port: 

```
mbed-os-example-cellular, Connecting...

Connection Established.
UDP: Sent 4 Bytes to echo.u-blox.com

Failure. Exiting

mbed-os-example-cellular, Connecting...

Connection Established.
UDP: Sent 4 Bytes to echo.u-blox.com
Received from echo server 4 Bytes

Success. Exiting
```
After a few presses of the reset button I always manage to get a response from the echo server.

This can be a bit tricky to reproduce as sometimes the board behaves as expected and sometimes I get stuck in the bug.  

## H2Compilers
I observed this malfunctioning with the 3 compilers

MarceloSalazar, 2017-12-06:
It seems related to this: https://github.com/ARMmbed/mbed-os-example-cellular/issues/31

gggekov, 2017-12-06:
Powering the board with a 5V usb power supply doesn't solve the issue - still getting some Failures on receiving reply from the echo server. Furthermore, the board stills connects slowly to the server.

MarceloSalazar, 2017-12-06:
@teetak01 please have a look

teetak01, 2017-12-07:
@AnttiKauppila

adbridge, 2017-12-07:
@AnttiKauppila Could you please look at this as a priority, we need to know if this is a genuine issue which could be a blocker , something that can be noted as a known issue or something that can be patched. If a fix needs to go into mbed-os for example that needs to be in, reviewed, through ci and merged ready for RC2 on Monday... Thanks

AnttiKauppila, 2017-12-08:
Under investigation

mhiljane, 2017-12-08:
Hi all, UDP is not reliable protocol and packets may get lost in the transmission process in internet. To get more reliable transmission/reception TCP should be used, which can be chosen from mbed_app.json by setting sock-type to be TCP.

MarceloSalazar, 2017-12-08:
Please don't close the ticket straight away without discussing and agreeing on next actions.

Conversations continues on Jira: IOTCELL-278

MarceloSalazar, 2017-12-12:
Waiting for PR (https://github.com/ARMmbed/mbed-os-example-cellular/pull/44) to get merged before closing.

MarceloSalazar, 2017-12-18:
Fixed

closed 2017-12-18 by MarceloSalazar