ARMmbed/mbed-os-example-cellular #41: Export project to Keil - device not present

gggekov, 2017-12-06:
I export the blinky to Keil via
`$ mbed export -i uvision -m UBLOX_C030_U201`

When opening the project in Keil I get to an error message "Missing device Family Pack STM32F4xx_DFS:2.11.0"
[image]

After clicking Import I am stuck at another error - "One or more Device Family Pack Devices are not present - would you like to choose another device"
[image]

What is the workaround to this problem? Is there some configuration in Keil that could solve the issue? 

Kind regards,
George

MarceloSalazar, 2017-12-06:
@theotherjimmy  please have a look

theotherjimmy, 2017-12-06:
Looks very similar to 
https://github.com/ARMmbed/mbed-os/issues/5467

gggekov, 2017-12-06:
I've tried the fix from ARMmbed/mbed-os#5467 but I get compile error messages after following the described steps.  Is there something else I could try to properly import the C030 as a target in Keil?

theotherjimmy, 2017-12-06:
@gggekov What compiler errors did you get?

gggekov, 2017-12-06:
Hi @theotherjimmy 

Here are the error messages

```
irq_cm4f.S: Error:  #5: cannot open source input file "__UVISION_VERSION SETA 524": No such file or directory
irq_cm4f.S: __UVISION_VERSION SETA 524: 0 warnings, 1 error
irq_cm4f.S: Error: A1067E: Output file specified as 'mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\TOOLCHAIN_ARM\TARGET_RTOS_M4_M7\irq_cm4f.S', but it has already been specified as '.\build\irq_cm4f.o'
irq_cm4f.S: error: A1905U: Pre-processor step failed for '__UVISION_VERSION SETA 524'

startup_stm32f437xx.S: Error:  #5: cannot open source input file "__UVISION_VERSION SETA 524": No such file or directory
startup_stm32f437xx.S: __UVISION_VERSION SETA 524: 0 warnings, 1 error
startup_stm32f437xx.S: Error: A1067E: Output file specified as 'mbed-os\targets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F437xG\device\TOOLCHAIN_ARM_STD\startup_stm32f437xx.S', but it has already been specified as '.\build\startup_stm32f437xx.o'
startup_stm32f437xx.S: error: A1905U: Pre-processor step failed for '__UVISION_VERSION SETA 524'
```

theotherjimmy, 2017-12-06:
@gggekov That's caused by not having the device support pack installed correctly. Have you installed it?

gggekov, 2017-12-06:
I've clicked the "install" button as per the first screenshot in the issue. However, after installing the missing packages I get stuck at the second screenshot from the description. 
Should I install all packages by hand?

theotherjimmy, 2017-12-08:
I think it may need a `variant`, which requires a large update to cmsis-pack-manager.

MarceloSalazar, 2017-12-11:
@adbridge I think we need to add this to the known-issue list

adbridge, 2017-12-12:
@theotherjimmy is there an outstanding issue to update the cmsis-pack-manager ? If not , can we have one please , then we can add something to the known issues if required, referencing that issue.

theotherjimmy, 2017-12-12:
@adbridge I created https://github.com/ARMmbed/cmsis-pack-manager/issues/16 to track the `variant` problem in cmsis-pack-manager.

adbridge, 2017-12-12:
@theotherjimmy thanks

ralphrmartin, 2018-02-10:
Any timescale on a fix for this? I also am stymied trying to export a project to uvision.

ralphrmartin, 2018-02-12:
Soory, could you give me some direct instructions on what to do, please? The oblique references above are rather mystifying...

theotherjimmy, 2018-02-12:
@ralphrmartin The text above claim that uvision complains with:

>"One or more Device Family Pack Devices are not present - would you like to choose another device"

You could choose the correct device at this point. I think that would at least allow you to build your code on devices with the "variant" problem.

ralphrmartin, 2018-02-12:
I already tried that, and I am afraid that doe not give me a solution which builds. I still get the errors shown.

theotherjimmy, 2018-02-12:
@ralphrmartin You could try replacing the string "STM32F437VG" with "STM32F437VGTx" in the to of your-project.uvprojx file. 
for example:
```xml 
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<Project xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="project_projx.xsd">

  <SchemaVersion>2.1</SchemaVersion>

  <Header>### uVision Project, (C) Keil Software</Header>

  <Targets>
    <Target>
      <TargetName>fat-fs</TargetName>
      <ToolsetNumber>0x4</ToolsetNumber>
      <ToolsetName>ARM-ADS</ToolsetName>
      <TargetOption>
        <TargetCommonOption>
          <Device>STM32F437VG</Device>
          <Vendor>STMicroelectronics</Vendor>
          <PackID>Keil.STM32F4xx_DFP.2.11.0</PackID>
          <PackURL>http://www.keil.com/pack</PackURL>
```
would become 
```xml
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<Project xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="project_projx.xsd">

  <SchemaVersion>2.1</SchemaVersion>

  <Header>### uVision Project, (C) Keil Software</Header>

  <Targets>
    <Target>
      <TargetName>fat-fs</TargetName>
      <ToolsetNumber>0x4</ToolsetNumber>
      <ToolsetName>ARM-ADS</ToolsetName>
      <TargetOption>
        <TargetCommonOption>
          <Device>STM32F437VGTx</Device>
          <Vendor>STMicroelectronics</Vendor>
          <PackID>Keil.STM32F4xx_DFP.2.11.0</PackID>
          <PackURL>http://www.keil.com/pack</PackURL>
```

theotherjimmy, 2018-02-12:
In case it was not clear: "STM32F437VGTx" is the only variant of the UBLOX_C030_U201

ralphrmartin, 2018-02-13:
Thanks, but in my case the device is a STM32F091RC. What replacement do I need in that case?

ralphrmartin, 2018-02-13:
Aha, OK, I had to change it to STM32F091RCTx. Thanks for the pointer which got me going.

theotherjimmy, 2018-02-13:
It's great to hear that you found the workaround you needed. I'm working on the verification of the variant parsing in cmsis-pack-manager. I'll try to keep you posted on my progress.

kimlep01, 2018-03-07:
I had the same error when exporting the example with MTS_DRAGONFLY_F411RE: "Device not found STM32F411RE". This would happen even after installing the required "Keil.STM32F4xx_DFP.2.12.0" pack.

The workaround: 
select STM32F411RETx as the device
use ST-Link debugger
Debug: connect with pre-reset
Instructions found here: https://github.com/ARMmbed/mbed-os/issues/5467

AnttiKauppila, 2018-03-08:
@theotherjimmy Any progress on this?

theotherjimmy, 2018-03-08:
Yep, but nothing releasable yet. I have variant parsing in cmsis-pack-manager working. I need to add a few more bits and then integrate it with mbed-os.

AriParkkila, 2018-06-04:
@theotherjimmy package update seems to work now with uVision V5.25.2.0 and Keil.STM32F4xx_DFP.2.13.0.

There is still a Warning #440: Requested device 'STM32F437VG' is subsituted with variant 'STM32F437VGTx' for target 'mbed-os-example-cellular'

I don't have actual HW to test if it really-really works :)

theotherjimmy, 2018-06-05:
@AriParkkila The difference between the Variants of that device are restricted to package differences. That should work.

AriParkkila, 2018-06-07:
@gggekov can you check if this issue is still valid?

gggekov, 2018-06-08:
I no longer have this problem, I export a project and am able to open it in Keil normally(without the error message in the first post). Therefore I close the issue.

closed 2018-06-08 by gggekov