ARMmbed/mbed-os-example-cellular #42: C030-U201 board not available to import into the online compiler

gggekov, 2017-12-06:
The C030-U201 board https://os-partnerbeta.mbed.com/platforms/ublox-C030-U201/ can't be imported in the online compiler. The button "Add to the online compiler" is missing on the page of the board. Therefore it's impossible to add it as a target in the online IDE.

adbridge, 2017-12-08:
@teetak01 @hasnainvirk  Is this board even officially supported yet then ?

JammuKekkonen, 2017-12-08:
Seems to be in web site at least. Also mbed-os supports the target. https://os.mbed.com/platforms/ublox-C030-U201/

adbridge, 2017-12-08:
@JammuKekkonen  You miss the point, on that supported platform page there is no option to import the target into the online IDE, which implies it isn't complete ?

MarceloSalazar, 2017-12-11:
I've discussed with @gggekov 
There is an (internal) way to import the platform on Online IDE.
Pending action for @gggekov to try and report feedback.

gggekov, 2017-12-11:
I confirm the solution of @MarceloSalazar and the application runs fine.

closed 2017-12-11 by gggekov