ARMmbed/mbed-os-example-cellular #43: Export to IAR in Windows -

gggekov, 2017-12-07:
I'm having problems when exporting the example application to IAR7 - "The system cannot find the file specified". The full error message and all the commands I'm using are below.  

```
C:\Users\geogek01\Documents\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.7.0-oob ≡ +5 ~1 -0 !]> mbed config --list
[mbed] Global config:
IAR_PATH=C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5\arm

[mbed] Local config (C:\Users\geogek01\Documents\mbedCLIproj\mbed-os-example-cellular):
TOOLCHAIN=GCC_ARM
TARGET=UBLOX_C030_U201
C:\Users\geogek01\Documents\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.7.0-oob ≡ +5 ~1 -0 !]> make clean
rmdir /S /Q "BUILD"
C:\Users\geogek01\Documents\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.7.0-oob ≡ +5 ~1 -0 !]> mbed export -i make_iar -m UBLOX_
C030_U201
Scan: .
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
C:\Users\geogek01\Documents\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.7.0-oob ≡ +5 ~1 -0 !]> make
C:/Users/geogek01/Documents/mbedCLIproj/mbed-os-example-cellular/makefile:781: warning: overriding commands for target `.s.o'
C:/Users/geogek01/Documents/mbedCLIproj/mbed-os-example-cellular/makefile:773: warning: ignoring old commands for target `.s.o'
"Compile: main.cpp"
process_begin: CreateProcess(NULL, iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 -On -r -DMBED_DEBUG -DMBE
D_TRAP_ERRORS_ENABLED=1 --enable_restrict --cpu Cortex-M4F --thumb --dlib_config DLib_Config_Full.h --c++ --no_rtti --no_exceptions
--guard_calls --no_static_destruction --guard_calls --no_static_destruction -DTARGET_STM32F437xx -DFEATURE_LWIP=1 -DTARGET_UBLOX_C03
0 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DDEVICE_PORTOUT=1 -DTARGET_STM32F437 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object
-DTARGET_STM32F4 -D__CMSIS_RTOS -DDEVICE_ANALOGOUT=1 -DMBEDTLS_CONFIG_HW_SUPPORT -DTARGET_UBLOX_C030_U201 -DDEVICE_I2C_ASYNCH=1 -DAR
M_MATH_CM4 -DTARGET_DEBUG -DTARGET_LIKE_CORTEX_M4 -DDEVICE_TRNG=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SPI_ASYNCH=1 -DTO
OLCHAIN_IAR -DTARGET_STM32F437xG -D__MBED__=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DRTC_LSI=1 -D__CORTEX_M
4 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_PWMOUT=1 -DTARGET_FAMILY_STM32 -DFEATURE_COMMON_PAL=1 -DDEVICE_PORTIN=1 -DHSE_VALUE=12000000 -D
TARGET_STM -DDEVICE_SERIAL=1 -DDEVICE_PORTINOUT=1 -DGNSSBAUD=9600 -DTARGET_LIKE_MBED -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGE
T_FF_ARDUINO -DDEVICE_SPI=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DTARGET_STM32F437VG -DMBED_BUILD_TIMESTA
MP=1512660702.48 -DDEVICE_FLASH=1 -DTARGET_CORTEX_M -DDEVICE_SERIAL_FC=1 --preinclude=mbed_config.h -I../ -I../. -I.././mbed-os -I..
/./mbed-os/cmsis -I.././mbed-os/cmsis/TARGET_CORTEX_M -I.././mbed-os/cmsis/TARGET_CORTEX_M/TOOLCHAIN_IAR -I.././mbed-os/drivers -I..
/./mbed-os/events -I.././mbed-os/events/equeue -I.././mbed-os/features -I.././mbed-os/features/FEATURE_COMMON_PAL -I.././mbed-os/fea
tures/FEATURE_COMMON_PAL/mbed-client-randlib -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-client-randlib/mbed-client-randlib -I..
/./mbed-os/features/FEATURE_COMMON_PAL/mbed-client-randlib/mbed-client-randlib/platform -I.././mbed-os/features/FEATURE_COMMON_PAL/m
bed-client-randlib/source -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-coap -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/
doxygen -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/mbed-coap -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/source -
I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/source/include -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-trace -I.././mbed
-os/features/FEATURE_COMMON_PAL/mbed-trace/mbed-trace -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-trace/source -I.././mbed-os/fe
atures/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/cs_
nvm -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/nvm -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostac
k-libservice -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/mbed-client-libservice -I.././mbed-os/features/FEATURE_
COMMON_PAL/nanostack-libservice/mbed-client-libservice/platform -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/sour
ce -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/IPv6_fcf_lib -I.././mbed-os/features/FEATURE_COMMON_PAL/na
nostack-libservice/source/libBits -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/libList -I.././mbed-os/feat
ures/FEATURE_COMMON_PAL/nanostack-libservice/source/libTrace -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/
libTrace/scripts -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/libip6string -I.././mbed-os/features/FEATURE
_COMMON_PAL/nanostack-libservice/source/nsdynmemLIB -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/nvmHelper
 -I.././mbed-os/features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop -I.././mbed-os/features/FEATURE_COMMON_PAL/sal-stack-nanos
tack-eventloop/nanostack-event-loop -I.././mbed-os/features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/nanostack-event-loop/pl
atform -I.././mbed-os/features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/source -I.././mbed-os/features/FEATURE_LWIP -I.././m
bed-os/features/FEATURE_LWIP/lwip-interface -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip -I.././mbed-os/features/FEATURE
_LWIP/lwip-interface/lwip-eth -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch -I.././mbed-os/features/FEATURE_LWIP
/lwip-interface/lwip-eth/arch/TARGET_STM -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/TARGET_STM32F4
 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/TARGET_STM32F4/TARGET_UBLOX_C030 -I.././mbed-os/featur
es/FEATURE_LWIP/lwip-interface/lwip-sys -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-sys/arch -I.././mbed-os/features/FE
ATURE_LWIP/lwip-interface/lwip/src -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/api -I.././mbed-os/features/FEATURE_
LWIP/lwip-interface/lwip/src/core -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/core/ipv4 -I.././mbed-os/features/FEA
TURE_LWIP/lwip-interface/lwip/src/core/ipv6 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include -I.././mbed-os/feat
ures/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip/pri
v -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip/prot -I.././mbed-os/features/FEATURE_LWIP/lwip-interface
/lwip/src/include/netif -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif/ppp -I.././mbed-os/features/FEATU
RE_LWIP/lwip-interface/lwip/src/include/netif/ppp/polarssl -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif -I../.
/mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif/ppp -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif/
ppp/polarssl -I.././mbed-os/features/filesystem -I.././mbed-os/features/filesystem/bd -I.././mbed-os/features/filesystem/fat -I.././
mbed-os/features/filesystem/fat/ChaN -I.././mbed-os/features/filesystem/littlefs -I.././mbed-os/features/filesystem/littlefs/TESTS_C
OMMON -I.././mbed-os/features/filesystem/littlefs/littlefs -I.././mbed-os/features/frameworks -I.././mbed-os/features/frameworks/gre
entea-client -I.././mbed-os/features/frameworks/greentea-client/greentea-client -I.././mbed-os/features/frameworks/greentea-client/s
ource -I.././mbed-os/features/frameworks/unity -I.././mbed-os/features/frameworks/unity/source -I.././mbed-os/features/frameworks/un
[... 4 lines cut ...]
s/mbedtls/targets/TARGET_STM -I.././mbed-os/features/mbedtls/targets/TARGET_STM/TARGET_STM32F4 -I.././mbed-os/features/mbedtls/targe
ts/TARGET_STM/TARGET_STM32F4/TARGET_STM32F437xG -I.././mbed-os/features/nanostack -I.././mbed-os/features/netsocket -I.././mbed-os/f
eatures/netsocket/cellular -I.././mbed-os/features/netsocket/cellular/generic_modem_driver -I.././mbed-os/features/netsocket/cellula
r/utils -I.././mbed-os/features/storage -I.././mbed-os/hal -I.././mbed-os/hal/storage_abstraction -I.././mbed-os/platform -I.././mbe
d-os/rtos -I.././mbed-os/rtos/TARGET_CORTEX -I.././mbed-os/rtos/TARGET_CORTEX/rtx4 -I.././mbed-os/rtos/TARGET_CORTEX/rtx5 -I.././mbe
[... cut ...]
TARGET_STM32F4/TARGET_STM32F437xG/device/TOOLCHAIN_IAR -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F4/device -o main.o ../main.cpp
, ...) failed.
make (e=2): The system cannot find the file specified.
make[1]: *** [main.o] Error 2
make: *** [all] Error 2
```

A research on google suggests it's probably an issue with my environmental variables.  I've tried to set up IAR through the system variables as well but the problem persists. 

Do you have any suggestions how to solve the problem? 

Thanks,

George

MarceloSalazar, 2017-12-07:
@gggekov can you please confirm the version of IAR you are using? It seems you are using 7.5.
Please do update to 7.8 and let us know how it works.

gggekov, 2017-12-08:
I'm using IAR 7.8 in the description of the issue.
```
C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5\arm\bin> .\iccarm.exe --version
IAR ANSI C/C++ Compiler V7.80.1.11864/W32 for ARM
```

MarceloSalazar, 2017-12-08:
@theotherjimmy please have look

theotherjimmy, 2017-12-08:
@gggekov You need to be able to execute `iccarm` from anywhere in your `cmd.exe`. You probably will need to add the location of `iccarm` to your PATH environment variable.

MarceloSalazar, 2017-12-08:
@theotherjimmy if the global settings were set (IAR_PATH), shouldn't this just work?
I'd expect the Makefile to be updated with the path to the toolchain. What do you think?

theotherjimmy, 2017-12-08:
@MarceloSalazar No. That's not how the exporters work. How would that work with the online compiler? it has no knowlage of your IAR_PATH environment var.

gggekov, 2017-12-11:
I confirm that adding the location of `iccarm` to the system Path solved the problem. The application is now working.

closed 2017-12-11 by gggekov