ARMmbed/mbed-os-example-cellular #54: OnboardModemInterface used incorrectly

kjbracey-arm, 2018-03-01:
This example invokes methods on `OnboardModemInterface` that are not part of the `CellularBase` abstract class. This means it may not work if `OnboardModemInterface` is not a `PPPCellularInterface`.

https://github.com/ARMmbed/mbed-os/pull/6082 has worked around this by adding the same extensions to its `EasyCellularInterface`, but that should not have been necessary.

This can probably be fixed for mbed OS 5.9 as part of shifting from `OnboardModemInterface` to `CellularBase::get_default_instance()` or `CellularDevice::get_default_instance()` - see https://github.com/ARMmbed/mbed-os/pull/6108

AriParkkila, 2018-06-04:
Example app uses only methods from CellularBase as there is no need to call `iface.modem_debug_on` anymore. That was changed to compile time definition `cellular.debug-at` in https://github.com/ARMmbed/mbed-os-example-cellular/blob/mbed-os-5.9.0-oob/mbed_app.json

AriParkkila, 2018-06-04:
@kjbracey-arm does the fix look like we could close this issue?

kjbracey-arm, 2018-06-04:
I still see the `modem_debug_on` in master.

If you just deleted that, it probably clears the issue for now, but it could easily recur as long as you're directly instantiating and using the object. It would be nice enforce the abstraction by using `CellularBase::get_default_instance()` (now in 5.9) and just a `CellularBase` pointer.

5.9.0 doesn't have a `CellularBase`-to-`CellularDevice` (or whatever) mapping API to check for advanced facilities. That's something else to look at at some point.

AriParkkila, 2018-06-04:
Ok, I see there are two problems to fix:
* Only use CellularBase methods (remove `modem_debug_on`)
* Instantiate `CellularBase::get_default_instance()` instead of `OnboardCellularInterface`

kjbracey-arm, 2018-06-04:
Yep. Doing the second ensures you've done the first.

AriParkkila, 2018-06-06:
@kjbracey-arm please check PR #83

AriParkkila, 2018-06-07:
Fix is now merged in the mbed-os-5.9.0-oob branch.

This issue can be closed when mbed-os-5.9.0-oob is merged to master.

AriParkkila, 2018-06-20:
@kjbracey-arm can we close this issue?

AriParkkila, 2018-06-26:
As there has been no further requests, I'm closing this issue now.

closed 2018-06-26 by AriParkkila