ARMmbed/mbed-os-example-cellular #59: APN and Auth Info not available

dhwalters423, 2018-03-16:
APN and auth info for the MTB Dragonfly is not available:

In the documentation
In the example
On the board page: https://os.mbed.com/modules/multitech-dragonfly/

This is a blocker, without this information the module is not usable with cellular.

Output when `apn` `username` and `password` is 0 in mbed_app.json (indicating none) returns the following error:

 `Couldn't connect: -3012, will retry`

AnttiKauppila, 2018-03-16:
ARM internal ref: IOTCELL-706

jarvte, 2018-03-19:
Hi,

APN and credentials are not Dragonfly specific but SIM card. 
APN is access point name specified by the operator which's SIM is in use.

https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-cellular/

In mbed_app.json file you need to change "sim-pin-code" to match you sim card pin code.
"apn" can be left empty if you don't know the right apn.

Most probably you don't need to change "username" and "password".

dhwalters423, 2018-03-19:
The Dragonfly MTB has an embedded SIM so I am thinking there must be a standard Apn. Would this come from the factory?

https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-cellular/

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

jarvte, 2018-03-20:
Talked with @dhwalters423 and there is no esim and this case can be closed.

closed 2018-03-20 by jarvte