ARMmbed/mbed-os-example-cellular #70: Dragonfly module not detected by mbed detect command

gggekov, 2018-05-31:
I've connected a Dragonfly module via Module Test Bed v1. However, the Dragonfly module is not detected by `mbed detect` command and I get the following error message:
```
C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.9.0-oob +5 ~0 -0 !]> mbed detect
Traceback (most recent call last):
  File "C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular\mbed-os\tools\detect_targets.py", line 82, in main
    if re.match(mcu_filter, mut['mcu']):
  File "c:\python27\lib\re.py", line 141, in match
    return _compile(pattern, flags).match(string)
TypeError: expected string or buffer
[ERROR] expected string or buffer
[mbed] ERROR: Failed to correctly detect all targets (run with '-vv' for more information)
```
My pip version is 10.0.1 and mbed-cli version is 1.5.1
```
C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.9.0-oob +5 ~0 -0 !]> mbed --version
1.5.1
C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.9.0-oob +5 ~0 -0 !]> pip --version
pip 10.0.1 from c:\python27\lib\site-packages\pip (python 2.7)
```

Can you please take a look into that problem? 
Thank you and kind regards,
George

MarceloSalazar, 2018-05-31:
@gggekov 

Can you run 'mbed-ls' and share the output?
Make sure the board has the latest [interface firmware](https://os.mbed.com/platforms/MTS-Dragonfly/#interface-firmware)

MarceloSalazar, 2018-05-31:
I didn't say this works with DAPLink ;-)
Please use the link provided to update the [interface firmware](https://os.mbed.com/platforms/MTS-Dragonfly/#interface-firmware)

ashok-rao, 2018-05-31:
@gggekov : looking at PinNames.h https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_MTB_MTS_DRAGONFLY/PinNames.h#L204  .. I see LED1 is already defined. You may have to upgrade your mbed-ls version. For the DAPLink, that MTB I gave you should have dragonfly DAPLink, I can send you the DAPLink binary again if needed..

gggekov, 2018-05-31:
Thank you @ashok-rao 
I updated the board with the image you gave me but am still getting the same result for mbed detect. What is the next step? Is the Dragonfly supposed to be supported by the mbed detect command?

ashok-rao, 2018-05-31:
mbedls should definitely work.. [from here](https://github.com/ARMmbed/mbed-ls/blob/master/mbed_lstools/platform_database.py#L95) .. I'm not sure about mbed detect..

adbridge, 2018-05-31:
@theotherjimmy  Do you know if the mbed detect command should work for this target ?

gggekov, 2018-06-01:
@theotherjimmy 
My `mbedls` is version 1.2.14 and for `mbed-ls` I am getting error(see my second post for the error message)

adbridge, 2018-06-01:
Latest version of mbed-ls is 1.4.5 @gggekov I suggest you upgrade

gggekov, 2018-06-01:
How can you do that?
`mbed-ls -update` and `mbed -update` do not work.

theotherjimmy, 2018-06-01:
@gggekov How did you get mbed-ls originally? Probably `pip install mbed-ls`. I recommend the same thing with the Upgrade flag: `pip install -U mbed-ls`.

gggekov, 2018-06-01:
Hi @theotherjimmy 
I must have used the Windows installed to get mbed-cli initially.
```
C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.9.0-oob +5 ~0 -0 !]> pip install
_U mbed-ls
ERROR: unknown command "install_U" - maybe you meant "install"
```

theotherjimmy, 2018-06-01:
@gggekov You typed `install_U` I wrote `install -U` change `_` to ` -` (space, dash).

gggekov, 2018-06-04:
In fact I had done only `pip install -U mbed-cli`
Updating `mbed-ls ` solved the problem and `mbed detect` works fine.

closed 2018-06-04 by gggekov