ARMmbed/mbed-os-example-cellular #71: Dragonfly error- Attempt to override undefined parameter 'platform.stdio-buffered-serial' in 'application[*]'

gggekov, 2018-05-31:
Compiling with ARM results in an error message: Attempt to override undefined parameter 'platform.stdio-buffered-serial' in 'application[*]'.
```
:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.9.0-oob +5 ~1 -0 !]> mbed compile -t ARM -m mts_dra
gonfly_f411re
Building project mbed-os-example-cellular (MTS_DRAGONFLY_F411RE, ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
[ERROR] Attempt to override undefined parameter 'platform.stdio-buffered-serial' in 'application[*]'
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular\mbed-os\tools
\make.py -t ARM -m mts_dragonfly_f411re --source . --build .\BUILD\mts_dragonfly_f411re\ARM" in "C:\Users\geogek01\Documents\Engineering\mb
edCLIproj\mbed-os-example-cellular"
---
```
How can this be solved? I am on the `mbed-os-5.9.0-oob` branch . The same error message appears with GCC_ARM and IAR.

AriParkkila, 2018-05-31:
Did you run _mbed update_ for mbed-os-5.9?

gggekov, 2018-05-31:
Yes, this solved the issue. Thank you!

closed 2018-05-31 by gggekov