ARMmbed/mbed-os-example-cellular #72: Example fails to compile for C030 in beta online comiler

dhwalters423, 2018-05-31:
When importing to the Beta online compiler: https://os-partnerbeta.mbed.com/compiler/ the example fails to compile with the following error:

[image]

On the live compiler: https://os.mbed.com/compiler/ the example compiles correctly and I am able to successfully run the example on the board.

gggekov, 2018-05-31:
I also encounter the same problem in https://os-partnerbeta.mbed.com.

dhwalters423, 2018-06-07:
I have tried again later (today 07 June) and was able to successfully compile. @theotherjimmy is there any action we can take to increase bandwidth with the beta compiler?

ashok-rao, 2018-06-07:
Thanks for re-checking this @dhwalters423 . Partner beta will undergo some maintenance after 5.9 OoB..such issues should be resolved after that. Issue can be closed now!

closed 2018-06-08 by adbridge