ARMmbed/mbed-os-example-cellular #74: modem_trace does not work unless mbed-trace is also enabled

dhwalters423, 2018-05-31:
Documentation lists **Turning modem AT echo trace on** and **Turning on the tracing and trace level** as 2 separate sections.

Modem trace only works when mbed-trace is also enabled.

Consider moving **Turning modem AT echo trace on** inside the  **Turning on the tracing and trace level** section, and adding a note that  mbed-trace is required to be enabled to see the modem-trace.

AriParkkila, 2018-06-01:
Fixed in changes:
* https://github.com/ARMmbed/mbed-os-example-cellular/pull/69
* https://github.com/ARMmbed/mbed-os/pull/7084

The fixes make AT-trace independent of mbed-trace flag.

ashok-rao, 2018-06-05:
Thanks @AriParkkila for the quick action. But, I think we should NOT enable AT trace "together" with mbed-trace. This will spit out lots of information (similiar to what happens in the nanostack/cloud client examples). I think it's better to keep these separate and give an option to the user to individually enable either AT or mbed-trace only or both depending on needs..
@dhwalters423 , @MarceloSalazar ok for you?

MarceloSalazar, 2018-06-05:
I understand each of the options have different purposes.
I'd suggest giving the option to the user to choose them independently.

dhwalters423, 2018-06-05:
I agree with @ashok-rao  and @MarceloSalazar approach here. Doesn't @AriParkkila 's PR accomplish that?

AriParkkila, 2018-06-06:
The PR above introduces two individual flags. The both can be enabled separately, that is you can have either one or the both enabled:
* `cellular.debug-at` for Turning modem AT echo trace on
* `mbed-trace.enable` for Turning on the tracing and trace level

So I think this work just the way you have requested above.

AriParkkila, 2018-06-07:
Thanks, this issue can be closed when mbed-os-5.9.0-oob is merged to master.

closed 2018-06-23 by dhwalters423