ARMmbed/mbed-os-example-cellular #75: Dragonfly module fails to get connected to the cellular network

gggekov, 2018-05-31:
I struggle to connect the Dragonfly module to the cellular network. I am trying both TCP and UDP sockets and on both occasions have the following output: 
```
mbed-os-example-cellular
Establishing connection
.....................................................................................................................................................

Couldn't connect: -3004, will retry
```

Here is my mbed_app.json file:
```
{
    "config": {
        "sock-type": "UDP",
        "modem_trace": {
            "help": "Turns AT command trace on/off from the cellular modem, defaults to off",
            "value": true
        },
        "sim-pin-code": {
            "help": "SIM PIN code",
            "value": 0
        },
        "apn": {
            "help": "The APN string to use for this SIM/network, set to 0 if none",
            "value": "\"comms365.o2.com\""
        },
        "username": {
            "help": "The user name string to use for this APN, set to zero if none",
            "value": "\"******\""
        },
        "password": {
            "help": "The password string to use for this APN, set to 0 if none",
            "value": "\"******\""
        },
        "trace-level": {
            "help": "Options are TRACE_LEVEL_ERROR,TRACE_LEVEL_WARN,TRACE_LEVEL_INFO,TRACE_LEVEL_DEBUG",
            "macro_name": "MBED_TRACE_MAX_LEVEL",
            "value": "TRACE_LEVEL_DEBUG"
        }
  },
    "target_overrides": {
        "*": {
            "target.features_add": ["LWIP", "COMMON_PAL"],
            "mbed-trace.enable": false,
            "lwip.ipv4-enabled": true,
            "lwip.ethernet-enabled": false,
            "lwip.ppp-enabled": true,
            "lwip.tcp-enabled": true,
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "platform.stdio-buffered-serial": true
        }
    }
}

```
In the json I compile I replace the * of username and password with the real values.  Also note that I have enabled trace debug and AT trace but don't get any debug information on the serial output.

I experience this problem with ARM compiler and in the online IDE. 

Do you have any suggestion what is wrong and how to fix it? 

Thank you and kind regards,
George

ashok-rao, 2018-05-31:
@hasnainvirk .. any thoughts?

AriParkkila, 2018-06-01:
Do you have SIM that requires PIN, such as

        "sim-pin-code": {
            "help": "SIM PIN code",
            "value": "1234"
        }

You need to enable trace flag to see more details:

        "mbed-trace.enable": true

You can see more AT debug prints when PR #69 is integrated.

gggekov, 2018-06-01:
Hi @AriParkkila 
Thank you for the comment. I attach the debug trace when I try to connect over TCP.
```
mbed-os-example-cellular
Establishing connection
[00000000ms][INFO][CELL]: CELLULAR_DEVICE: TELIT_HE910
[00000001ms][DBG ][CELL]: process_oob readable=0, pos=0, len=0
[00000001ms][DBG ][CELL]: process_oob exit
[00000002ms][INFO][CELL]: Init state, waiting 0 ms before POWER state)
[00000008ms][INFO][CELL]: Cellular state from Init to Power
[00000015ms][INFO][CELL]: cellular_status: Init ==> Power
[00000021ms][INFO][CELL]: Cellular power ON (timeout 1000 ms)
[00000328ms][INFO][CELL]: Cellular state from Power to Device ready
[00000328ms][INFO][CELL]: cellular_status: Power ==> Device ready
[00001330ms][WARN][CELL]: AT timeout
[00001330ms][DBG ][CELL]: consume_to_tag not found
[00001330ms][WARN][CELL]: AT stop tag not found
[00001351ms][WARN][CELL]: AT timeout
[00001351ms][DBG ][CELL]: consume_to_tag not found
[00001351ms][WARN][CELL]: AT stop tag not found
[00001352ms][DBG ][CELL]: Retry State Device ready, retry 1/10
[00001352ms][INFO][CELL]: Cellular event in 2 seconds
ATE0ATE0

#STN: 5,0600
[00004359ms][WARN][CELL]: AT timeout
[00004359ms][DBG ][CELL]: consume_to_tag not found
[00004359ms][WARN][CELL]: AT stop tag not found
[00004380ms][WARN][CELL]: AT timeout
[00004380ms][DBG ][CELL]: consume_to_tag not found
[00004380ms][WARN][CELL]: AT stop tag not found
[00004381ms][DBG ][CELL]: Retry State Device ready, retry 2/10
[00004381ms][INFO][CELL]: Cellular event in 4 seconds
ATE0ATE0
OK
AT+CMEE=1
OK
AT&K0;&C1;&D0
OK
[00008471ms][INFO][CELL]: Cellular device ready
AT+CGREG=2
OK
AT+CREG=2
OK
[00008540ms][INFO][CELL]: Cellular state from Device ready to SIM pin
[00008540ms][INFO][CELL]: cellular_status: Device ready ==> SIM pin
[00008541ms][INFO][CELL]: Sim state (timeout 1000 ms)
AT+CPIN?
+CPIN: READY

OK
[00008576ms][INFO][CELL]: SIM is ready
[00008576ms][INFO][CELL]: Cellular state from SIM pin to Registering network
[00008577ms][INFO][CELL]: cellular_status: SIM pin ==> Registering network
AT+CGREG?
+CGREG: 2,0

OK
[00008612ms][DBG ][CELL]: get_network_registration: type=1, status=0
AT+CREG?
+CREG: 2,2

OK
[00008648ms][DBG ][CELL]: get_network_registration: type=2, status=2
AT+COPS?
+COPS: 0

OK
[00008683ms][DBG ][CELL]: automatic registering mode: 0
[00008683ms][DBG ][CELL]: Retry State Registering network, retry 1/10
[... 365 lines cut ...]
+CREG: 0
[00165479ms][DBG ][CELL]: urc_creg
[00165479ms][WARN][CELL]: not registered
[00165480ms][DBG ][CELL]: stat: 0, lac: -1, cellID: -1, act: -1
[00165480ms][INFO][CELL]: FSM: network_callback called with event: 4098, intptr: 0, _state: Registering network
[00165490ms][DBG ][CELL]: process_oob exit
[00170476ms][DBG ][CELL]: process_oob readable=1, pos=0, len=0

+CREG: 2
[00170479ms][DBG ][CELL]: urc_creg
[00170479ms][DBG ][CELL]: stat: 2, lac: -1, cellID: -1, act: -1
[00170480ms][INFO][CELL]: FSM: network_callback called with event: 4098, intptr: 2, _state: Registering network
[00170485ms][DBG ][CELL]: process_oob exit
[00170676ms][DBG ][CELL]: process_oob readable=1, pos=0, len=0

+CREG: 0
[00170679ms][DBG ][CELL]: urc_creg
[00170679ms][WARN][CELL]: not registered
[00170680ms][DBG ][CELL]: stat: 0, lac: -1, cellID: -1, act: -1
[00170680ms][INFO][CELL]: FSM: network_callback called with event: 4098, intptr: 0, _state: Registering network
[00170690ms][DBG ][CELL]: process_oob exit
```
It looks like it fails to register to the cellular network. Do you have an idea what may the reason for that? 
Thank you,
George

AriParkkila, 2018-06-01:
The modem is searching for a new operator, but it can't find a suitable network to register. Could you check (from the operator) that the SIM has also GPRS enabled for a data network?

gggekov, 2018-06-01:
How can I do this check?
When I put the SIM into a mobile phone, it connects straight away to O2 network.

AriParkkila, 2018-06-04:
I guess that your phone should display 3G data connection icon when GPRS is active? You could also ask that from your SIM operator too.

One explanation could be that you have e.g. O2/comms365 4G SIM without 3G fallback? Because your phone most likely supports 4G, it can connect to 4G network. But Dragonfly has only 3G support so it can't connect to network because SIM has only 4G enabled.

AriParkkila, 2018-06-05:
@gggekov "The SIM card has working 4G connection"...  Your Dragonfly modem only supports 3G connection. Could you check that SIM card also have 3G connectivity?

gggekov, 2018-06-08:
Hi @AriParkkila 
I confirm the SIM card has 3G connection. 
Running the example on the `mbed-os-5.9.0-rc2` branch results in the same output as before. I am getting repeatedly the following message. 
```
+CREG: 0
[00126528ms][DBG ][CELL]: urc_creg
[00126528ms][WARN][CELL]: not registered
[00126529ms][DBG ][CELL]: stat: 0, lac: -1, cellID: -1, act: -1
[00126529ms][INFO][CELL]: FSM: network_callback called with event: 4098, intptr: 0, _state: Registering network
[00126539ms][DBG ][CELL]: process_oob exit
```

AriParkkila, 2018-06-08:
+CREG: 2 means that modem is searching for an GSM operator.
+CREG: 0 means that modem is **not** searching for an GSM operator.

It's fine that modem sends CREG: 0/2 notifications in a loop when the modem is trying to find a suitable network. The problem is that the log has no +CGREG notifications, which would indicate that modem tries to find also a 3G/GPRS network.

Can you insert the SIM in your phone and search for available networks to see that there is O2 network with 3G service. Then you could try to set your phone in 3G only mode (disable 4G) to see that your phone can connect to that 3G network.

gggekov, 2018-06-08:
Hi Ari,
Yes, I confirm you that when I insert the SIM card in a phone it connects to the O2 network and my phone has data. I've tried with two different SIM cards and I am getting the same error message.

AriParkkila, 2018-06-08:
@gggekov did you disable 4G on your phone?

gggekov, 2018-06-08:
Hi @AriParkkila 
I confirm you I have set network mode as 3G-only on my phone and I do have internet connection with the same SIM card I am using on the Dragonfly.
If no one knows who to fix this problem at the moment, probably you may enable it for the next release of MbedOS.

AriParkkila, 2018-06-11:
@gggekov can you use successfully any SW with that Dragonfly board + SIM combination? Could this issue be because of broken HW or old firmware?

gggekov, 2018-06-11:
Hello Ari,
The mbed-os-example-cellular is the only test I've done that is combining the Dragonfly board + SIM card.  Please let me know  if there is another example app I can try that combines SIM card + Dragonfly. The Dragonfly board itself is working fine. 
Kind regards,
George

AriParkkila, 2018-06-11:
Yes, it would be good to check if modem firmware could be updated.

You could also try https://github.com/ARMmbed/mbed-os-example-cellular/tree/mbed-os-5.6-oob, which has previously been verified to work with Dragonfly.

SeppoTakalo, 2018-06-11:
If this already works on one of the tester, close this one and continue debugging network related issues on other thread.
This problem sound more generic to Cellular APIs and networking, than this application.

AriParkkila, 2018-06-18:
@gggekov any updates to this issue?

Do you think we could close this Github issue as @SeppoTakalo suggested, and continue debugging in ARM Internal Ref: IOTCELL-1037.

closed 2018-06-18 by gggekov