ARMmbed/mbed-os-example-cellular #76: Troubleshooting: Suggestion for documentation improvement

gggekov, 2018-06-01:
In the Troubleshooting section of https://github.com/ARMmbed/mbed-os-example-cellular/ provide information what to do when the cellular board doesn't connect to the network and the user is getting the following error message: 
```
mbed-os-example-cellular
Establishing connection
.....................................................................................................................................................

Couldn't connect: -3004, will retry
```
Suggested information to be included:

- Make sure the fields `sim-pin-code`, `apn`, `username` and `password` from the `mbed_app.json` file are filled in correctly. The correct values should appear in the user manual of the board if using eSIM or in the details of the SIM card if using normal SIM.
- Enable trace flag to have access to debug information `"mbed-trace.enable": true`
- Try both `TCP` and `UDP` socket types 
- Add other engineering tips

AriParkkila, 2018-06-04:
@gggekov thanks a lot, I updated troubleshooting in
https://github.com/ARMmbed/mbed-os-example-cellular/pull/69

AriParkkila, 2018-06-04:
The change is now merged in OOB branch. See Troubleshooting chapter in https://github.com/ARMmbed/mbed-os-example-cellular/tree/mbed-os-5.9.0-oob

Please close this issue, if the update looks good to you.

closed 2018-06-04 by gggekov