ARMmbed/mbed-os-example-cellular #81: Exporters don't work for Dragonfly

gggekov, 2018-06-05:
The `make_armc5` and `make_gcc_arm` exporters don't work properly for the Dragonfly board.
The error message I am getting is:
```
C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.9.0-oob +10 ~1 -0 !]> mbed export -i make_gcc_arm -
m MTB_MTS_DRAGONFLY
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S [{matrix,ides}]] [-E] [--build BUILD_DIR]
                  [--source SOURCE_DIR] [-D MACROS] [--profile PROFILE]
                  [--update-packs] [--app-config APP_CONFIG] [--ignore IGNORE]
project.py: error: MTB_MTS_DRAGONFLY not supported by make_gcc_arm
[mbed] ERROR: "c:\python27\python.exe" returned error code 2.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular\mbed-os\tools
\project.py -i make_gcc_arm -m MTB_MTS_DRAGONFLY --source ." in "C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellul
ar"
---
C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular [mbed-os-5.9.0-oob +10 ~1 -0 !]> mbed export -i make_armc5 -m
MTB_MTS_DRAGONFLY
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S [{matrix,ides}]] [-E] [--build BUILD_DIR]
                  [--source SOURCE_DIR] [-D MACROS] [--profile PROFILE]
                  [--update-packs] [--app-config APP_CONFIG] [--ignore IGNORE]
project.py: error: MTB_MTS_DRAGONFLY not supported by make_armc5
[mbed] ERROR: "c:\python27\python.exe" returned error code 2.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular\mbed-os\tools
\project.py -i make_armc5 -m MTB_MTS_DRAGONFLY --source ." in "C:\Users\geogek01\Documents\Engineering\mbedCLIproj\mbed-os-example-cellular
"
```

AriParkkila, 2018-06-06:
`mbed export -S` shows that only `iar` is supported for MTB_MTS_DRAGONFLY.

Maybe you could open an issue to mbed-cli to enhance `export` so that it tells explicitly which exports are supported for the current target instead of telling just which is not supported..?

gggekov, 2018-06-06:
Thank you Ari for the comment. 
I close the issue since you confirm Dragonfly is not supported by make_armc5 and make_gcc_arm.

closed 2018-06-06 by gggekov