ARMmbed/mbed-os-example-cellular #86: mbed assertation failed: iface, file: ./main.cpp, line 228

dlfryar, 2018-06-18:
Out of the box build results in mbed assertation failed: iface, file: ./main.cpp, line 228

Updated mbed_app.json to add trace and debug

Updated main.cpp to add a 20 second wait after the dot_thread.start() which results in a HardFault.

Tried with the following hardware:

- [Gemalto Cinterion Connect Shield](https://www.gemalto.com/brochures-site/download-site/Documents/m2m-connect-shield.pdf)
- [NXP FRDM-K64F](https://os.mbed.com/platforms/FRDM-K64F/)
- [Avnet AT&T WNC 14A2A Cellular IoT Kit](https://os.mbed.com/components/Avnet-ATT-WNC-14A2A-Cellular-IoT-Kit/)

Gemalto direct AT configuration

```
OK
ati1
Cinterion
ELS61-US
REVISION 01.000
A-REVISION 00.026.00
 
OK
at+cops=?
+COPS: (2,"AT&T","AT&T","310410",7),(1,"AT&T","AT&T","310410",2),(1,"","","313100",7),(1,"T-Mobile","T-Mobile","310260",2))
 
OK
at+cops?
+COPS: 0,0,"AT&T",7
 
OK
at+creg?
+CREG: 0,1
 
OK
at+cgdcont?
+CGDCONT: 1,"IP","m2m.com.attz.mnc170.mcc310.gprs","",0,0
 
+CGDCONT: 3,"IP","BROADBAND","",0,0
 
OK
```

[main.cpp.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/2113134/main.cpp.txt)
[mbed_app.json.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/2113135/mbed_app.json.txt)
[capture-2018-06-18-001.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/2113131/capture-2018-06-18-001.txt)

dlfryar, 2018-06-18:
cc: @AriParkkila

AriParkkila, 2018-06-19:
@dlfryar thanks for good logs.

1) Assert is expected as K64F does not have _hosted cellular module_. You seem to be using K64F + Gemalto ELS61 module so you need to add this in the `mbed_app.json`:

    "macros": ["CELLULAR_DEVICE=GEMALTO_ELS61", "MDMRXD=D0", "MDMTXD=D1"]

2) You need to define also `"mbed-trace.enable": true` or update to _mbed-os-master_ because we are missing some debug commits on the _mbed-os-5.9_ branch.

3) Example seems to have a bug in `dot_event()` which does not check that `iface` is created before using it. We need to fix that.

dlfryar, 2018-06-19:
Here's another run with the recommended changes:

##### Copy Gemalto drivers from [ciot-example](https://github.com/ARMmbed/ciot-example)

cp -R ciot-example/source/cellular/targets/GEMALTO mbed-os-example-cellular/
[mbed_app.json.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/2115744/mbed_app.json.txt)
mbed-os/features/cellular/framework/targets

##### Build syntax

```
cd mbed-os && git pull && mbed update master && cd ..
mbed compile -t GCC_ARM -m K64F
```

##### Error output

```
mbed-os-example-cellular
Establishing connection

.

++ MbedOS Error Info ++
Error Status: 0x80010130 Code: 304 Module: 1
Error Message: could not pinout
Location: 0x183a7
Error Value: 0x1009
Current Thread: Id: 0x2000da0c Entry: 0x18ee1 StackSize: 0x1000 StackMem: 0x2000da58 SP: 0x2000e988 
-- MbedOS Error Info --

mbed-os-example-cellular
Establishing connection

.

++ MbedOS Error Info ++
Error Status: 0x80010130 Code: 304 Module: 1
Error Message: could not pinout
Location: 0x183a7
Error Value: 0x1009
Current Thread: Id: 0x2000da0c Entry: 0x18ee1 StackSize: 0x1000 StackMem: 0x2000da58 SP: 0x2000e988 
-- MbedOS Error Info --
```

dlfryar, 2018-06-19:
One additional datapoint is that I have not been able to validate hardware flow control working with the Gemalto module.  Configuring "MDMRTS=D2" and "MDMCTS=D3" does not seem to work as expected.  I'm still having to jumper D2 and D3 in order for the board to respond on the AT interface.

AriParkkila, 2018-06-20:
@dlfryar there seems to be a NULL pointer crash if APN lookup is enabled. We will fix this immediately. ARM internal ref IOTCELL-1122

I don't actually know which pins should be defined on [K64F](https://os.mbed.com/platforms/FRDM-K64F/) as RTS/CTS?

AriParkkila, 2018-06-25:
@dlfryar I think this can be closed as the both crash fixes are now merged, please see:
 * https://github.com/ARMmbed/mbed-os/pull/7275
* https://github.com/ARMmbed/mbed-os-example-cellular/commit/ecf43d8fffd0c9448d6755e9891f54890532873a

I guess you need to bridge RTS/CTS pins on Cinterion just the way you have already done.

dlfryar, 2018-06-25:
```
mbed-o

++ MbedOS Error Info ++
Error Status: 0x80010130 Code: 304 Module: 1
Error Message: could not pinout
Location: 0x181d3
Error Value: 0x1009
Current Thread: Id: 0x2000da0c Entry: 0x18d0d StackSize: 0x1000 StackMem: 0x2000da58 SP: 0x2000e988 
-- MbedOS Error Info --
```

Reran with the same mbed_app.json above and a fresh clone including the copy step for the Gemalto drivers.

[build.log](https://github.com/ARMmbed/mbed-os-example-cellular/files/2133633/build.log)

dlfryar, 2018-06-25:
I'll work out the RTS/CTS issue once the example builds and runs cleanly.  I did learn that the Gemalto Cinterion module has the CTS/RTS pins fixed at D2/D3 so bending the pins and jumper wiring it is necessary.

#### FRDM-K64F

PTC16 – RX – D0
PTC17 – TX – D1
PTC18 – RTS – J6 pin 8
PTB9 – CTS – D2

[image]

dlfryar, 2018-06-25:
@AriParkkila can you test with the K64F + Cinterion shield then with the NRF52840_DK?

AriParkkila, 2018-06-26:
@dlfryar thanks for the pin map. That seems to work, but note that you need to use K64F pin PTC18 in configuration:
    "macros": ["CELLULAR_DEVICE=GEMALTO_ELS61", "MDMRXD=D0", "MDMTXD=D1", "MDMRTS=PTC18", "MDMCTS=D2"]

I tried that on both Cinterion EVB and Connect Shield. It seems that my connect shield does not need RTS/CTS to be bridged at all, but it's enough to connect TX/RX.

Also NRF52840_DK EVB seems to work fine with RS232 shield (without RTS/CTS) when wired to Connect Shield:
    "macros": ["CELLULAR_DEVICE=GEMALTO_ELS61", "MDMRXD=D0", "MDMTXD=D1"]

dlfryar, 2018-06-26:
Verified assertion is resolved with latest release.

closed 2018-06-26 by dlfryar