ARMmbed/mbed-os-example-cellular #87: RadioAccessTechnology does not have a configuration parameter in mbed-os config system

dlfryar, 2018-06-18:
RadioAccessTechnology is not configurable using the mbed-os config system.

Configuration example:

   "radio-access-technology": {
      "help": "GSM,GSM_COMPACT,UTRAN,EGPRS,HSDPA,HSUPA,HSDPA_HSUPA,E_UTRAN,CATM1,NB1,UNKNOWN",
      "value": "\"CATM1\""     
    },

https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/API/CellularNetwork.h#L119

Same related question in the [forum](https://os.mbed.com/questions/81051/Change-Radio-Access-Technology-in-EasyCe/)

dlfryar, 2018-06-18:
cc: @AriParkkila

AriParkkila, 2018-06-19:
@dlfryar thanks, we will add that.

AnttiKauppila, 2018-09-12:
@AriParkkila What is the status of this?

kivaisan, 2018-10-01:
I don't think RAT should be a cellular library configuration as RAT can be changed via CellularNetwork's `nsapi_error_t set_access_technology(RadioAccessTechnology rat)` API.

If user is using EasyCellularConnection, this can be called for example like:
```
EasyCellularConnection cellConn(true);
cellConn.get_device()->open_network(cellConn.get_serial())->set_access_technology(RAT_NB1);
```

AriParkkila, 2018-11-26:
PR https://github.com/ARMmbed/mbed-os/pull/8821

closed 2018-12-12 by AriParkkila