ARMmbed/mbed-os-example-cellular #125: Hard-fault exception when compiling with IAR v8.32.1 (mbed-os-5.12.0-rc2) OoB testing

chuanga, 2019-03-19:
Mbed OS version: mbed-os-5.12.0-rc2
Platform: MTB_ADV_WISE_1570
IAR: v8.32.1 - https://netstorage.iar.com/SuppDB/Protected/PRODUPD/013356/EWARM-CD-8321-18631.exe 
       v7.5 is working fine.
Error log:
==============================================
-- MbedOS Error Info --

mbed-os-example-cellular

Built: Mar 19 2019, 11:07:25

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 00000018
R1   : 00020026
R2   : 20001838
R3   : 00000000
R4   : FFFFFFFF
R5   : 20001798
R6   : 200017D8
R7   : FFFFFFFF
R8   : 0CCCCCCB
R9   : 0800261A
R10  : 00000000
R11  : 00000000
R12  : 08001C5D
SP   : 20001780
LR   : 08009145
PC   : 080017B4
xPSR : 01000000
PSP  : 20001760
MSP  : 10000510
CPUID: 410FC241
HFSR : 80000000
MMFSR: 00000000
BFSR : 00000000
UFSR : 00000000
DFSR : 00000002
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x0
Error Value: 0x80017B4
Current Thread: main  Id: 0x20002A2C Entry: 0x80019F5 StackSize: 0x1000 StackMem: 0x20000878 SP: 0x100004B8
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=MTB_ADV_WISE_1570
-- MbedOS Error Info --

kivaisan, 2019-03-20:
IAR 8.32 does not seem to allow null string as printf parameter. Fix PRs created.

adbridge, 2019-03-21:
This should now be fixed and the fix come into RC4

kivaisan, 2019-03-21:
I merged the new PR for 5.12 branch https://github.com/ARMmbed/mbed-os-example-cellular/pull/128

closed 2019-03-27 by adbridge