ARMmbed/mbed-os-example-cellular #132: Need example updated to Mbed OS 5.12+

MarceloSalazar, 2019-04-09:
Please update mbed-os.lib and required configuration (mbed_app.json) for multiple platforms/shields to use this example as a reference for Mbed OS 5.12 based applications.
Thanks!

MarceloSalazar, 2019-04-09:
Related to https://github.com/ARMmbed/mbed-os-example-cellular/pull/131

AnttiKauppila, 2019-04-09:
@MarceloSalazar Which platforms are you referring?

MarceloSalazar, 2019-04-09:
From my point of view, is not a problem about platform, but rather configuration to enable a shield using serial/SPI (as I understand there is a new way of doing this in 5.12).
Can be K64 or L475 platforms with both BG96 or Gemalto shields.

Once this new config is in place, we can introduce more for other shields that are currently in the works.

AnttiKauppila, 2019-04-09:
@AriParkkila Can you help with this?

AriParkkila, 2019-04-11:
@MarceloSalazar It's now updated to Mbed OS 5.12+. Can we close this ticket?

MarceloSalazar, 2019-04-11:
Yes, thank you!

closed 2019-04-11 by MarceloSalazar