ARMmbed/mbed-os-example-cellular #137: Not working on SDT64B

heejungPark, 2019-05-02:
Hi, I tested this code on our device(SDT64B).
But it is not connected to server and I got a log([Log.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/3137022/Log.txt)).
I don't know what the problem is.
Could you please check this log and answer me?
Thanks.
Here are my code(main.cpp, mbed_app.json).
[mbed-os-example-cellular.zip](https://github.com/ARMmbed/mbed-os-example-cellular/files/3137028/mbed-os-example-cellular.zip)

heejungPark, 2019-05-02:
I found one thing.
I modify PIN number from "1234" to "0000".
But same result...

heejungPark, 2019-05-02:
For more detail log, "cellular.debug-at" is changed to true.
Here is the log([Log2.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/3137688/Log2.txt)).

mirelachirica, 2019-05-02:
@heejungPark 
Logs are showing that there are 2 PDP contexts defined:
+CGD CONT: 1,"IPV6","lte-internet..sktelecom.com","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 2,"IP" ,"lte-internet.sktelecom.com","0.0.0.0",0,0,0,0

This might cause the problem.

In order to delete existing PDP contexts you could try adding following "one time" routine to your application, before connect:

```
ATHandler *at = CellularDevice::get_default_instance()->get_at_handler();
    for (int i=0; i<7; i++) {
        at->cmd_start("AT+CGDCONT=");
        at->write_int(i);
        at->cmd_stop();
        at->resp_start();
        at->resp_stop();
        at->clear_error();
    }
 CellularDevice::get_default_instance()->release_at_handler(at);

```
Let us know if that is helping the case or not.

heejungPark, 2019-05-03:
@mirelachirica 

For reference, we are using BG96 as Cat.M1 module.
With this module, we additionally tested two projects except this code(`mbed-os-example-cellular`).
1. https://os.mbed.com/users/hkjung/code/WIZnet-IoTShield-BG96-Ping => Success
1.  https://os.mbed.com/users/hkjung/code/mbed-os-example-cellular-BG96 => Fail

Because of the result of first project, I think BG96 module could be normal.
Actually I refer to the second project for setting mbed_app.json of `mbed-os-example-cellular`.
May I know whether this json file has a problem?
And, as you mentioned, I add that code before calling connect(). But I got a almost same results.
Please confirm this log and source code ([file.zip](https://github.com/ARMmbed/mbed-os-example-cellular/files/3142047/file.zip)).

Thanks.

mirelachirica, 2019-05-03:
> May I know whether this json file has a problem?

Can you try removing "target.features_add": ["LWIP"] from json file?

heejungPark, 2019-05-06:
I tried it but same results...

mirelachirica, 2019-05-06:
I just noticed in the log file that the clearing of contexts before connect wasn't working since the modem was not ready to receive AT commands yet. So it has to be done later:
- somewhere in AT cellular stack for example in begining of AT_CellularContext::get_context()
OR 
- define cellular callback in main that would check for SIM READY state and do the routine of clearing the contexts at that point:

```
MBED_ASSERT(iface);
iface->attach(cellular_callback);
```

```
void cellular_callback(nsapi_event_t ev, intptr_t ptr)
{
    // support for testing in non-blocking mode and step by step
    if (ev >= NSAPI_EVENT_CELLULAR_STATUS_BASE && ev <= NSAPI_EVENT_CELLULAR_STATUS_END) {
        cell_callback_data_t *ptr_data = (cell_callback_data_t *)ptr;
        cellular_connection_status_t cell_ev = (cellular_connection_status_t)ev;
        if (cell_ev == CellularSIMStatusChanged && ptr_data->error == NSAPI_ERROR_OK &&
            ptr_data->status_data == CellularDevice::SimStateReady) {

                ATHandler *at = CellularDevice::get_default_instance()->get_at_handler();
                for (int i=0; i<7; i++) {
                    at->cmd_start("AT+CGDCONT=");
                    at->write_int(i);
                    at->cmd_stop();
                    at->resp_start();
                    at->resp_stop();
                    at->clear_error();
                }
                CellularDevice::get_default_instance()->release_at_handler(at);
        }
    }
}
```
Reminder: this should be done/needed only ONCE.

heejungPark, 2019-05-06:
I used your second way on my project.
The result is same but a little message is different.
Here are my files. Please refer it.
[File.zip](https://github.com/ARMmbed/mbed-os-example-cellular/files/3146856/File.zip)

mirelachirica, 2019-05-06:
Now from log it can be seen that old context were deleted correctly.
But QIACT still fails. From WIZnet-IoTShield-BG96-Ping example I see the context is setup with QICSGP not the standard CGDCONT. So what could be tried is replace in AT_CellularContext::set_new_context(...)
```
    _at.cmd_start("AT+CGDCONT=");
    _at.write_int(cid);
    _at.write_string(pdp_type_str);
    _at.write_string(_apn);
```
with

   ```
  _at.cmd_start("AT+QICSGP=");
    _at.write_int(cid);
    _at.write_int(2); //ipv6 in WIZnet-IoTShield-BG96-Ping, 1 would be value for ipv4
    _at.write_string(_apn);
    _at.write_string("");
    _at.write_string("");
    _at.write_int(0);
```

heejungPark, 2019-05-07:
Yes, it makes a difference.
Now I can see "Connection Established." message.
But the next step could be fail.
Here are the logs([Log5.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/3150477/Log5.txt)).
Thanks.

heejungPark, 2019-05-07:
I have a additional question.
Why are the old contexts from WIZnet-IoTShield-BG96-Ping example left?
To set this module to initial state, is there a way that I delete all contexts in module?
Thanks.

mirelachirica, 2019-05-07:
> But the next step could be fail.
> Here are the logs([Log5.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/3150477/Log5.txt)).

The DNS name resoving fails. You could try one of the following:
- to see if basic test to send and receive from echo server is working:
**in json file change "\"echo.mbedcloudtesting.com\"" to "\"52.215.34.155\"" -> no DNS resolving will be needed**

- in current log the IP address for the modem is IPv6 address: 32.1.2.216.19.172.90.185.0.0.0.0.105.136.157.0.  
To see if it helps the DNS resolving if IP address is IPV4:
**do the AT+QICSGP with context type IPV4: _at.write_int(2); -> _at.write_int(1);**

mirelachirica, 2019-05-07:
> To set this module to initial state, is there a way that I delete all contexts in module?

Other than suggested above no. There is no API at the moment to do this.

heejungPark, 2019-05-07:
Is it right as follows?
Please check this([Log6.txt](https://github.com/ARMmbed/mbed-os-example-.cellular/files/3152058/Log6.txt)).
Thanks.
[image]
[image]

mirelachirica, 2019-05-07:
I cannot access the Log6.txt file.

mirelachirica, 2019-05-07:
> Is it right as follows?

Yes, the modifications look correct. It can be tried with both 1 and 2 values for QICSGP, once the echo server is set in json to be the IP address.

heejungPark, 2019-05-07:
Sorry please access again?
[Log6.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/3152224/Log6.txt)

mirelachirica, 2019-05-07:
Log6 shows again 2 contexts. Can you enable the clearing of contexts code?

heejungPark, 2019-05-07:
Oh sorry, I attached log again.
[Log7.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/3152338/Log7.txt)

mirelachirica, 2019-05-07:
Now old contexts are deleted but the context setup with QICSGP and type 1(IPv4) it fails to activate with QIACT. So please change back QICSGP to use type 2(IPv4v6) and leave the echo test as IP address.

heejungPark, 2019-05-08:
I changed code as follows.

[image]

[image]

So this try successes.
Here is the log.
[Log8.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/3155151/Log8.txt).

mirelachirica, 2019-05-08:
Good!

So it seems the DNS resolving doesnt work for some reason. From the log when DNS solving is taking place, I can see that there is no responses coming from any of the DNS servers where requests are sent: 8.8.8.8, 209.244.0.3, 84.200.69.80.

Is it possible that you could try with WIZnet-IoTShield-BG96-Ping example to call getIpAddressByName_BG96(...), to see if there is some generic problem of comunication with these DNS servers, in the cellular network you are conncted to?

heejungPark, 2019-05-08:
Sorry, it is not the answer about your request.

First, I checked a array named pdp_type_str[].
This array is always "IP".
I don't know why `modem_supports_ipv4` is true and `modem_supports_ipv6` is false?

For test, when I set "IPV6" or "IPV4V6" by force, it is working well as follows.
So, I think it could be another solution if user can control `modem_supports_ipv6` to true.
Is there a way that user can control that variable?

[image]

[image]

mirelachirica, 2019-05-08:
The IPv6 support for the modem has to be set in:
QUECTEL_BG96.cpp: cellular_properties[AT_CellularBase::PROPERTY_MAX]
...
0,  // PROPERTY_IPV6_STACK -> 1,  // PROPERTY_IPV6_STACK
...

mirelachirica, 2019-05-08:
> For test, when I set "IPV6" or "IPV4V6" by force, it is working well as follows.

What works well? The context activation with CGDCONT, so no need for QICSGP?

heejungPark, 2019-05-08:
Ah, I see, I found that.
Then, is it not care that I change value of this `cellular_properties`?

"IPV6" and "IPV4V6", both is working well.

mirelachirica, 2019-05-08:
> Then, is it not care that I change value of this `cellular_properties`?
No problem. It should be changed according with the modem support.

> "IPV6" and "IPV4V6", both is working well.
What works well, the context activation without need for QICSGP?

The echo server is still set as IP address?

heejungPark, 2019-05-08:
You can see it in a part of code and serial log, I used CGDCONT.
PROPERTY_IPV6_STACK of cellular_properties is set 1, it could be no problem to use CGDCONT.

And echo server is still set as IP address(52.215.34.155).
when echo.mbedcloudtesting.com, it is not working.

mirelachirica, 2019-05-08:
Ok, very good finding. Please let me know if you have time to check the DNS resolving as I proposed earlier. Thanks.

heejungPark, 2019-05-09:
Sorry but I don't know what I should input parameters of getIpAddressByName_BG96(...).

mirelachirica, 2019-05-09:
First parameter should be the name to be resolved: "echo.mbedcloudtesting.com" and second parameter should be a buffer where the IP address is saved:
```
char ipstr[50];
getIpAddressByName_BG96("echo.mbedcloudtesting.com", ipstr);
```

loverdeg-ep, 2019-05-14:
@mirelachirica Any further comments on the log provided?

mirelachirica, 2019-05-15:
Last log proves that in your network/setup, DNS resolving works when proprietary command AT+QIDNSGIP is used. We have to check and consider having support for this command.

heejungPark, 2019-05-15:
Okay thanks. 
It seems that USIM card of this module may be discarded.
If it is true, could it be a reason of our problem?

mirelachirica, 2019-05-21:
@heejungPark can you check if changing in QUECTEL_BG96_CellularStack::create_socket_impl(...): _at.write_string("127.0.0.1"); -> _at.write_string("0:0:0:0:0:0:0:1"); helps the DNS resolving?

This seems to be the case in following PR:
https://github.com/ARMmbed/mbed-os/pull/10473

DanielDmlee, 2019-05-21:
Hi @mirelachirica, Already @heejungPark and I had talked about this patch. We guess this problem was from an expired SIM card. As far as I know, heejungPark try to changing SIM card via seller.

AriParkkila, 2019-09-23:
Closing as resolved.

closed 2019-09-23 by AriParkkila