ARMmbed/mbed-os-example-cellular #147: Missing steps to working with NRF52840

hugueskamba, 2019-06-20:
The pins of the shield need to be specified in `mbed_app.json`. This information is missing on the documentation.

```
"target_overrides": {
        "*": {
            ...
            "QUECTEL_BG96.provide-default": true,
            "QUECTEL_BG96.tx": "p42",
            "QUECTEL_BG96.rx": "p40"
            ...
```

kivaisan, 2019-06-24:
@hugueskamba PIN settings are HW specific, so no generic settings can be put into example.
If you have some target which has known pins, those can be configured in driver's mbed_lib.json. For example BG96: https://github.com/ARMmbed/mbed-os/blob/15b5b5da23b970d397e7d94bfcda7e38fa3870c8/features/cellular/framework/targets/QUECTEL/BG96/mbed_lib.json#L41

hugueskamba, 2019-06-24:
> @hugueskamba PIN settings are HW specific, so no generic settings can be put into example.
> If you have some target which has known pins, those can be configured in driver's mbed_lib.json. For [...]

What I meant was that the documentation should make clear to the user that s/he might need to setup the PINs.

kivaisan, 2019-06-24:
@hugueskamba Ok. We'll check the documentation for this.

AriParkkila, 2019-09-20:
@hugueskamba This is documented in `NetworkInterface::get_default_instance()`:
```
     * The interface returned should be already configured for use such that its
     * connect() method works with no parameters. For connection types needing
     * configuration, settings should normally be obtained from JSON.
```
BG96 specific configuration is documented in its `mbed_lib.json`:
```
{
    "name": "QUECTEL_BG96",
    "config": {
        "tx": {
            "help": "TX pin for serial connection. D1 assumed if Arduino Form Factor, needs to be set/overwritten otherwise.",
            "value": null
        },
        "provide-default": {
            "help": "Provide as default CellularDevice [true/false]",
            "value": false
        }
    }
```

hugueskamba, 2019-09-20:
> @hugueskamba This is documented in `NetworkInterface::get_default_instance()`:
>
> ```
>      * The interface returned should be already configured for use such that its
>      * connect() method works [...]

Hi @AriParkkila ,

When doing OOB I follow the content of the README. Without that information in the README it does not work. I think it would be a good idea to add it to the README.

AriParkkila, 2019-09-20:
@hugueskamba That sounds good, fixed README at PR #161.

closed 2019-09-20 by AriParkkila