ARMmbed/mbed-os-example-cellular #172: Operator new[] out of memory with U-Blox C027

peksaa01, 2019-12-13:
Out of memory when U-Blox C027 compiled with ARMC6.13 or IAR 8.32.1. Trace on with level TRACE_LEVEL_DEBUG and debug-at true.

++ MbedOS Error Info ++
Error Status: 0x8001011F Code: 287 Module: 1
Error Message: Operator new[] out of memory
Location: 0x0
Error Value: 0x330
Current Thread: main Id: 0x10003264 Entry: 0xA009 StackSize: 0x1000 StackMem: 0x100009D8 SP: 0x100017F0
For more info, visit: https://mbed.com/s/error?error=0x8001011F&tgt=UBLOX_C027
-- MbedOS Error Info --

adbridge, 2019-12-13:
@AnttiKauppila @JanneKiiskila @jamesbeyond are you guys aware of any issues like this with the U-Blox board ?

JanneKiiskila, 2019-12-13:
Nope.

jamesbeyond, 2019-12-13:
No, I am not aware of this issue, if providing more info, e.g. which version of mbed-OS , (SHA would be even better). what steps to reproduce the issue, ( e.g. what mbed configuration/profile been used, what is the command line to build, on which example) that would be helpful to let us reproduce the issue.

In general, on a memory restraint platform using debug profile, could lead to out of memory.

adbridge, 2019-12-13:
@jamesbeyond this is with RC2 and the mbed-os-example-cellular

kimlep01, 2019-12-16:
Hi,

This is a known issue with the C027. We are not prioritising this board anymore due to RAM constraints and we suggest using UBLOX_C030_U201 which is a higher priority board for us.

nadaanas1988, 2020-10-04:
Hello

I'm facing the same issue. I'm using the U blox c027 and in my program I'm defining the threads as the following:

Thread GNSS_UDP(osPriorityNormal, 2000); //sending to the FMS the GNSS data
Thread GNSS_PB_UDP(osPriorityNormal, 2000);  // Sending the Push button 1 with the GNSS data to the truck
Thread TCP_Server(osPriorityNormal, 2000); //TCP server thread FMS
Thread Temp_Sensor(osPriorityNormal, 2000); //TCP server thread FMS
//test threads
Thread Safety_Relay(osPriorityNormal, 2000);
Thread UDP_Socket(osPriorityNormal, 2000);

The UDP socket thread always running without a sleep.  just triggering it with signals:

`void Safety_Relay_Thread()
{
    printf("Safety \n\r");
    //Inputs:
    int Local_SB1_O3_press = 0;
    int Local_SB2_O3_press = 0;
    int Local_SK_S34_press_old = 0;
    int Local_SK_S34_press_new;
    int Local_SK_14_press = 0;
    int Local_SK_24_press = 0;
    int Local_SK_32_press =0;   
    
    const char* SB1_O3_Status = "OFF";
    const char* SB2_O3_Status = "OFF";
    const char* SK_S34_Status = "OFF";
    const char* SK_14_Status = "OFF";
    const char* SK_24_Status = "OFF";
    const char* SK_32_Status = "OFF";
    
    bool UDP_sending;
    
    
    while(1)
    {
        Local_SB1_O3_press = SB1_O3;
        Local_SB2_O3_press = SB2_O3;
        Local_SK_S34_press_new = SK_S34;
        Local_SK_14_press = SK_14;
        Local_SK_24_press = SK_24;
        Local_SK_32_press = SK_32;
        
        SB1_O3_Mutex.lock () ;
        SB1_O3_press = Local_SB1_O3_press;
        SB1_O3_Mutex.unlock();

        SB2_O3_Mutex.lock () ;
        SB2_O3_press = Local_SB2_O3_press;
        SB2_O3_Mutex.unlock();
        
        SK_S34_Mutex.lock () ;
        SK_S34_press_new = Local_SK_S34_press_new;
        SK_S34_Mutex.unlock();
        
        SK_14_Mutex.lock () ;
        SK_14_press = Local_SK_14_press;
        SK_14_Mutex.unlock();
        
        SK_24_Mutex.lock () ;
        SK_24_press = Local_SK_24_press;
        SK_24_Mutex.unlock();
        
        SK_32_Mutex.lock () ;
        SK_32_press = Local_SK_32_press;
        SK_32_Mutex.unlock();
        
         if(Local_SB1_O3_press ==0)
[... 151 lines cut ...]
        {
            strcpy(buffer, "Emergency");
            //myled_01=0;
        }
        
        Network_Mutex.lock();
        int ret = td_sock.sendto(td_addr, buffer, strlen(buffer));
        Network_Mutex.unlock();
        td_sock.close();
        
        //ThisThread::sleep_for(1);
    }  
}`

Please your advise.

The error msg is :

 ++ MbedOS Error Info ++
Error Status: 0x8001011F Code: 287 Module: 1
 Error Message: Operator new[] out of memory
Location: 0xB96F
 Error Value: 0x28
 Current Thread: application_unnamed_thread  Id: 0x10000BF8 Entry: 0xA871 StackSize: 0x7D0 StackMem: 0x100067C8 SP: 0x10006E74For more info, visit: https://mbed.com/s/error?error=0x8001011F&tgt=UBLOX_C027
                                                                             -- MbedOS Error Info --

paul-szczepanek-arm, 2021-01-04:
There's not much that can be done with a memory constraint other than to use a dev board with more memory. The debug profile will use more memory. Short of restructuring your program to not use as many threads all I can suggest is running with --release or maybe trying the baremetal profile `requires: ["baremetal"]`.

closed 2021-01-04 by paul-szczepanek-arm