ARMmbed/mbed-os-example-cellular #177: P-L496G-CELL02 (DISCO_L496AG+BG96) does not ramp up to the network

soleilplanet, 2020-01-08:
I'm trying to get the P-L496G-CELL02 device to the network (using STModCellular, BG96), but the modem cannot attach to the network.
So I enabled AT logs and saw that the driver gets the RSSI value, but keeps sending 'AT_CEREG?' and gets the response of '"2,4", indicating that it can not register to the operator.
However, with the same SIM card, it ramps up to the network with another BC95G device and another BC28 device.
Logs attached:
[20200108_cellular-example.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4033749/20200108_cellular-example.txt)

soleilplanet, 2020-01-08:
@MarceloSalazar would you please help? Thanks!

jeromecoutant, 2020-01-08:
Hi
Maybe you could enable mbed-trace in order to check what happened in STModCellular ?

https://github.com/ARMmbed/mbed-os/tree/master/features/frameworks/mbed-trace

LMESTM, 2020-01-08:
Hmm ... 
Make sure the Antenna is well mounted. 
Make sure you're selecting the proper SIM slot (there is embedded sim and external sim on this module
Also be patient, it may take a pretty long time (several minutes, up to 10?) for the module to scan all the supported frequency range the first time it connects to the NW

soleilplanet, 2020-01-08:
attaching the logs with mbed-trace set to debug level:
[20200108_cellular-example-02.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4034343/20200108_cellular-example-02.txt)
The signal strength shown in the log is the same with the results showing with other NB-IoT devices. (it's not bad, around -63db - -57db)
Thanks for helping!

AnttiKauppila, 2020-01-08:
Would it be possible to get logs also from success case?

mirelachirica, 2020-01-08:
At least logs show APN lookup used:

[00010529ms][INFO][CELL]: Looked up APN internet

Could you try define the correct APN in json:

"nsapi.default-cellular-apn": "\"your-apn\""

soleilplanet, 2020-01-09:
@mirelachirica  I've set APN lookup to false with the operator's APN, logs attached:
[20200109_cellular-example-02.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4038689/20200109_cellular-example-02.txt)
However, for your information, APN lookup works on other devices with R410M-02B and BC95G with this operator.
@AnttiKauppila  unfortunately this module has never successfully ramped up to the NB-IoT network, so I don't have the log for a successful case at the moment.
@LMESTM  is there any chance I can have the modem firmware of the module upgraded? (I connected the micro USB socket on the module to PC to see if I can use the Quectel Q-Navigator to check the status, but PC does not recognize the device at all)

AnttiKauppila, 2020-01-09:
@soleilplanet I was meaning logs with other modems )R410M-02B and BC95G) so we can see if the logs differ with failing BG96 from SIM perspective. I was thinking exactly the same as Mirela pointed out with APN lookup.

soleilplanet, 2020-01-09:
@AnttiKauppila  I just collected the log with BC95G (apn-lookup is true):
[20200109_bc95g_ref_log.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4039202/20200109_bc95g_ref_log.txt)
I just noticed that I forgot to attach the antenna to this device so the RSSI shown in the log is -109, but somehow it still ramps up to the network.

LMESTM, 2020-01-09:
@soleilplanet 

You can find information in the X-CUBE-Cellular package (you can follow the release note information)
https://my.st.com/content/my_st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32cube-expansion-packages/x-cube-cellular.html

In this package, there is a utility with some information on how to flash the modem:
\Utilities\Modem_FW\BG96\Pack_BG96MAR02A08M1G_01.012.01.012.zip

mirelachirica, 2020-01-09:
> keeps sending 'AT_CEREG?' and gets the response of '"2,4"

"2,4" is response to AT+CGREG(GPRS network registration status)
response for AT+CEREG(EPS network registration status ) is "2,2" -> searching for network + last one is "2,0" meaning is not searching anymore
A succesful run with BG96 looks as follows: 
[BG96_succesfull_EPS_nw_registration.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4040279/BG96_succesfull_EPS_nw_registration.txt)

No different than your logs other than RSSI signal which is stronger in your case -61 dBm vs. -83dBm.

Two things could still be checked:
- band configuration AT+QCFG=“band” -> If the last value(band for NB1) of the three returned values is not 0xA0E189F (CM_BAND_PREF_ANY) then correct value should be set.
- flag "cellular.clear-on-connect": true

soleilplanet, 2020-01-10:
@mirelachirica  I checked the command and it returns 0xf,0x80,0x80, so I think it's using band 8 for NB, which is the band this operator is using.
However, I tried to set it to CM_BAND_PREF_ANY, but keeps seeing errors; would you please help to see if I'm calling the right way?
	printf("sending (%s) to modem...", "AT+QCFG=\"band\",0xf,0x80,0xA0E189F,0\n");
	_at.flush(); // flush existing data
	_at.lock();
	_at.cmd_start("AT+QCFG=\"band\",0xf,0x80,0xA0E189F,0");
	_at.cmd_stop();
	_at.set_delimiter(0);
	_at.resp_start("+QCFG:");
	_at.read_string(buff, sizeof(buff));
        _at.resp_stop();
        _at.set_default_delimiter();
and the response I get: 
[00009876ms][DBG ][CELL]: AT flush
[00009882ms][INFO][CELL]: AT TX (47): AT+QCFG="band", 0xf, 0x400A0E189F, 0xA0E189F, 1
[00009890ms][INFO][CELL]: AT TX ( 1): <cr>
[00009895ms][INFO][CELL]: AT RX ( 9): <cr><ln>ERROR<cr><ln>
[00009902ms][DBG ][CELL]: AT error -3012

I got the same error when I call this way:
_at.at_cmd_discard("+AT+QCFG", "=", "%s%d%d%d%d", "\"band\"", 0xf, 0x80, 0xA0E189F, 1);

soleilplanet, 2020-01-10:
@LMESTM @mirelachirica I've upgraded to the new firmware; unfortunately it doesn't seem to help:
[20200110_band_check_new_bg96_fw_002.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4044362/20200110_band_check_new_bg96_fw_002.txt)
upgraded to version: BG96MAR02A08M1G

mirelachirica, 2020-01-10:
@soleilplanet for some reason this command succeeds only when CR is feed between arguments instead of comma ','.
Following format is returning OK for me:

_at.at_cmd_discard("+QCFG","=", "%s%b%b%b%d", "band", "0xf",3,"0x80",3,"0xa0e189f",9,1);

soleilplanet, 2020-01-13:
@mirelachirica  I tried the command you shared and it is also returning OK here, but looks like it didn't take effect and the value of the configuration stays 0xf,0x80,0x80. Logs attached:
[20200113_cellular-example-01.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4052346/20200113_cellular-example-01.txt)
I've tried to set the <effect> field to be 0 and reboot the device, but it is still keeping the band 8 configuration, not CM_BAND_PREF_ANY.
Also, when <effect> field is set to 1 (as shown in the log), the config was still band 8, not CM_BAND_PREF_ANY.
@LMESTM is there anything I can check to see why the modem is not saving the config? 
Thanks!

LMESTM, 2020-01-13:
@soleilplanet  No clue sorry. My next recommendation would be to try the X-CUBE-CELLULAR SW package from st.com with it documentation and see if that can help. I think that this package offers a menu over console for setting up the bands.

mirelachirica, 2020-01-13:
> @mirelachirica I tried the command you shared and it is also returning OK here, but looks like it didn't take effect and the value of the configuration stays 0xf,0x80,0x80

@soleilplanet this format has effect:
   ```
 _at.cmd_start("AT+QCFG=\"band\",");
  _at.write_string("0", false); // 0 -> existing band value doesnt change
  _at.write_string("0", false); // 0 -> existing band value doesnt change
  _at.write_string("a0e189f", false);
  _at.write_int(1);
  _at.cmd_stop_read_resp();
```

soleilplanet, 2020-01-14:
@mirelachirica the command works and now NB-IoT band support has been changed to CM_BAND_PREF_ANY.
I tried multiple SIM cards from different operators but none of them work. Logs attached:
[20200114_cellular-example-cht-band8-01.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4057548/20200114_cellular-example-cht-band8-01.txt)
[20200114_cellular-example-fet-band28-01.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4057549/20200114_cellular-example-fet-band28-01.txt)
[20200114_cellular-example-stream-roaming-01.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4057550/20200114_cellular-example-stream-roaming-01.txt)
[20200114_cellular-example-twm-band20-01.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4057551/20200114_cellular-example-twm-band20-01.txt)
As you can see in the file name, there there are 4 operators (actually 3), band 8 for Chunghwa Telecom (CHT), band 20 for Taiwan Mobile (TWM), band 28 for FarEast Telecom (FET), and band 8 for Stream connectivity service (it should be running on CHT)
However, in the FET case (band 28), I see that the device has registered to the network and it tried to attach to the network, and CGATT reports that it is attached to the network, but it just got stuck there.

mirelachirica, 2020-01-16:
@soleilplanet you could also try manual registration by providing your operator PLMN in json: "nsapi.default-cellular-plmn": "\"mccmnc\""

Bband setting commands can be removed too since CM_BAND_PREF_ANY is already set for NB-IoT.

soleilplanet, 2020-01-16:
@mirelachirica I've tried to add PLMN earlier today, but it didn't help either. However, I managed to try another BG96 device (WIO BG96) and it works (either with or without PLMN). I am planning to try locking bands on 8, 20 and 28 and lock the mode to NB1 tomorrow and see how it goes. Will keep you posted.

soleilplanet, 2020-01-17:
@mirelachirica  @LMESTM 
I've locked the band to 8,20,28 and mode to NB1, but will not working.
This is the log with mode locked at NB1 and bands locked at 8,20,28, and PLMN is null:
[20200117_lockbandmode_test_003-null-plmn-rat9.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4075228/20200117_lockbandmode_test_003-null-plmn-rat9.txt)

However, during the test I found that the modem is actually not accepting AT+COPS=1,2,<plmn> and AT+COPS=1,2,<plmn>,<act>, it just always gives error.
[20200117_lockbandmode_test_002-plmn-46692-rat9-custom.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4075227/20200117_lockbandmode_test_002-plmn-46692-rat9-custom.txt)
As you can see in the log, both commands always receive -3012 error. 
And, in the log you can also see in the log that during the process there was once I sent "ATAT+COPS=1,2,plmn,act", and in that case the modem is returning "NO CARRIER" (line 371 to 378)
Any thoughts?
Thanks!

mirelachirica, 2020-01-21:
Manual COPS needs bigger AT timeout. It is not used or configurable at the moment. You can try in

` AT_CellularNetwork::set_registration(const char *plmn)` to replace:

```
if (_op_act != RAT_UNKNOWN) {
    return _at.at_cmd_discard("+COPS", "=", "%d%d%s%d", mode, format, plmn, _op_act);
} else {
    return _at.at_cmd_discard("+COPS", "=", "%d%d%s", mode, format, plmn);
}
```

with

```
_at.set_at_timeout(20*1000); //20 seconds
nsapi_error_t ret = NSAPI_ERROR_OK;
if (_op_act != RAT_UNKNOWN) {
    ret = _at.at_cmd_discard("+COPS", "=", "%d%d%s%d", mode, format, plmn, _op_act);
 } else {
    ret = _at.at_cmd_discard("+COPS", "=", "%d%d%s", mode, format, plmn);
 }
 _at.restore_at_timeout();
return ret;
```

soleilplanet, 2020-01-22:
@mirelachirica  I followed your modification, and now it's getting a CME ERROR: 30.
[20200122_manual_cops_001.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/4095860/20200122_manual_cops_001.txt)
There is a set of new device coming in 1-2 weeks, I guess we'll need to see if it's the same with the new one? Thanks!

mirelachirica, 2020-01-22:
> now it's getting a CME ERROR: 30

description for this: "No network service"

soleilplanet, 2020-02-05:
I've got the new set of L496AG-CELL02 and STMOD BG96 device and it works, so I think this would be a hardware issue.
@mirelachirica do you think you need the problematic board so see if there is any potential issue?
If not, I think we can close this issue. If so, please let me know.
Thanks!

jeromecoutant, 2020-02-05:
Hi
Note I have just added FW version log in https://github.com/ARMmbed/mbed-os/pull/12339
You could check the difference ?

soleilplanet, 2020-02-07:
@jeromecoutant  sure, but I am in self-quarantine now due to our coronavirus policy so I dont have access to the problematic board at the moment. I'll check the difference when I have access to the board.

jeromecoutant, 2020-02-07:
Second question/assumption :-)

Has someone already used the "faulty" modem before using it in mbed-os application ?
See for example https://github.com/ARMmbed/mbed-os/issues/12381 or
https://github.com/ARMmbed/mbed-os/pull/12340

Your new modem is maybe the same as the first one, but it starts from "reset" SW.

@AriParkkila

AriParkkila, 2020-02-17:
It's quite common here (as well) to see "faulty" BG96s, which actually just were written accidentally with some persistent configurations :) It may help to recover, if you flash a new/different firmware on the BG96.

@soleilplanet This issue can be closed?

jeromecoutant, 2020-04-29:
Let's close this issue ?
Thx

jeromecoutant, 2020-06-08:
ping

closed 2020-06-08 by soleilplanet