ARMmbed/mbed-os-example-cellular #217: pin map not found on peripheral

adityathakekar, 2021-03-08:
I am using dragino Quectel  nb-iot board with stm32 nucleo l496zg. 

- editing the mbed.json file for tx and rx pins according to the datasheet, i get this error.
- checked all the possible methods of connecting with TCP or UDP.
-  The board works perfectly with AT commands. 
What am i missing here?

jeromecoutant, 2021-03-08:
Hi
You should also provide your mbed_app.json file information.

adityathakekar, 2021-03-08:
{
    "config": {
        "sock-type": "UDP",
        "echo-server-hostname": {
            "help": "Echo server host name.",
            "value": "\"udp.connectivity-suite.cloud\""
        },
        "echo-server-port": {
            "help": "Echo server port number.",
            "value": 4445
        },
        "trace-level": {
            "help": "Options are TRACE_LEVEL_ERROR,TRACE_LEVEL_WARN,TRACE_LEVEL_INFO,TRACE_LEVEL_DEBUG",
            "macro_name": "MBED_TRACE_MAX_LEVEL",
            "value": "TRACE_LEVEL_INFO"
        }
    },
    "target_overrides": {
        "*": {
            "target.network-default-interface-type": "CELLULAR",
            "mbed-trace.enable": true,
            "lwip.ipv4-enabled": false,
            "lwip.ipv6-enabled": true,
            "lwip.ethernet-enabled": false,
            "lwip.ppp-enabled": false,
            "lwip.tcp-enabled": true,
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "platform.stdio-buffered-serial": true,
            "cellular.debug-at": false,
            "nsapi.default-cellular-plmn": null,
            "nsapi.default-cellular-sim-pin": null,
            "nsapi.default-cellular-apn": "\"iot.1nce.net\"",
            "nsapi.default-cellular-username": null,
            "nsapi.default-cellular-password": null,
            "QUECTEL_BG96.tx": "D11",
            "QUECTEL_BG96.rx": "D10",
            "QUECTEL_BG96.provide-default": true
        }
    }
}

jeromecoutant, 2021-03-08:
OK, I am not a BG96 expert, but I think QUECTEL_BG96.tx and rx pins are supoosed to be UART ?

See https://os.mbed.com/platforms/ST-Nucleo-L496ZG/#zio-and-arduino-compatible-headers
Change to D1 and D0

adityathakekar, 2021-03-08:
yes, those are supposed to be uart. 
step : 
1. Changed it to D0 and D1. 
2. module does not connect to the cellular even after doing this. 
3. pin map not found on peripheral error seems to disappear.

jeromecoutant, 2021-03-08:
Thx

Please open then a new issue with the new problem :-)

adityathakekar, 2021-03-08:
Thanks for your time.

closed 2021-03-08 by adityathakekar