ARMmbed/mbed-os-example-cellular #218: BG96 cellular does connection problem

adityathakekar, 2021-03-08:
changed the uart pins and edited the json as per the documentation. 
complete link to the previous issue can be found here.

https://github.com/ARMmbed/mbed-os-example-cellular/issues/217#issue-824624940

1. Cellular nb iot sim cannot connect to the network. 
2. we can connect to the nb-iot sim via arduino. 

Do we need AT commands in the code?
Or we just need to edit the json?

adityathakekar, 2021-03-09:
> @adityathakekar This issue has an incomplete or old issue template.For future reference please use an up to date clone of the repository before raising issues. Many thanks.

Hey, i am using up to date clone of repository and the code does not work ,connecting to bg96 cellular nb iot using udp.
Changed the uart pins to D1 and D0.

paul-szczepanek-arm, 2021-03-09:
Apologies for the bots, they can't be reasoned with.

I don't have your board unfortunately so you'll have to help me out a bit. Can you please enable traces, set them to debug level and share the log?

You can also try turning on AT prints with cellular.debug-at.

adityathakekar, 2021-03-09:
Modem power on
Connecting to cellular context in 0ms

and the timer continues. 
There is no connection with cellular network.

paul-szczepanek-arm, 2021-03-09:
I'd expect a bit more prints if you had tracing on.

adityathakekar, 2021-03-10:
23:18:09.448 -> 
[2K[33m[00011627ms][WARN][CELL]: URC already added with prefix: +CGEV: NW DEACT[0m
23:18:09.448 -> 
[2K[33m[00011627ms][WARN][CELL]: URC already added with prefix: +CGEV: ME DEACT[0m
23:18:09.448 -> 
[2K[33m[00011628ms][WARN][CELL]: URC already added with prefix: +CGEV: NW PDN D[0m
23:18:09.448 -> 
[2K[33m[00011628ms][WARN][CELL]: URC already added with prefix: +CGEV: ME PDN D[0m
23:18:09.495 -> 
[2K[33m[00011636ms][WARN][CELL]: URC already added with prefix: +QIURC: "pdpde[0m
23:18:09.495 -> 
[2K[39m[00011644ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:18:09.495 -> 
[2K[39m[00011649ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:18:10.432 -> 
23:18:10.432 -> mbed-os-cellular
23:18:10.432 -> 
[2K[39m[00000000ms][INFO][CELL]: New CellularContext  (0x20004E50)[0m
23:18:10.432 -> Establishing connection
23:18:10.432 -> 
[2K[39m[00000001ms][INFO][CELL]: CellularContext connect[0m
23:18:10.432 -> 
[2K[39m[00000002ms][INFO][CELL]: Start connecting (timeout 1000 ms)[0m
23:18:10.432 -> 
[2K[39m[00000002ms][INFO][CELL]: AT TX ( 2): AT[0m
23:18:10.432 -> 
[2K[39m[00000006ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:18:11.417 -> 
[2K[39m[00001003ms][INFO][CELL]: AT TX ( 2): AT[0m
23:18:11.417 -> 
[2K[39m[00001003ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:18:12.401 -> 
[2K[39m[00002004ms][INFO][CELL]: Modem power ON (timeout 1000 ms)[0m
23:18:12.448 -> 
[2K[39m[00002004ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:18:12.448 -> 
[2K[39m[00002005ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:18:13.526 -> 
[2K[39m[00003105ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:18:13.526 -> 
[2K[39m[00003105ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:18:14.604 -> 
[2K[39m[00004206ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:18:14.651 -> 
[2K[39m[00004206ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:18:15.729 -> 
[2K[39m[00005307ms][INFO][CELL]: Continue after 1 seconds[0m
23:18:16.714 -> 
[2K[33m[00006307ms][WARN][CELL]: URC already added with prefix: +CGEV: NW DEACT[0m
23:18:16.714 -> 
[2K[33m[00006307ms][WARN][CELL]: URC already added with prefix: +CGEV: ME DEACT[0m
23:18:16.760 -> 
[2K[33m[00006308ms][WARN][CELL]: URC already added with prefix: +CGEV: NW PDN D[0m
23:18:16.760 -> 
[2K[33m[00006308ms][WARN][CELL]: URC already added with prefix: +CGEV: ME PDN D[0m
23:18:16.760 -> 
[2K[33m[00006316ms][WARN][CELL]: URC already added with prefix: +QIURC: "pdpde[0m
23:18:16.760 -> 
[2K[39m[00006324ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:18:16.760 -> 
[2K[39m[00006329ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:18:17.839 -> 
[2K[39m[00007425ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:18:17.839 -> 
[2K[39m[00007425ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:18:18.964 -> 
[2K[39m[00008526ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:18:18.964 -> 
[2K[39m[00008526ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:18:20.042 -> 
[... 120 lines cut ...]
[2K[39m[00088227ms][INFO][CELL]: Continue after 64 seconds[0m
23:20:42.618 -> 
[2K[33m[00152227ms][WARN][CELL]: URC already added with prefix: +CGEV: NW DEACT[0m
23:20:42.665 -> 
[2K[33m[00152227ms][WARN][CELL]: URC already added with prefix: +CGEV: ME DEACT[0m
23:20:42.665 -> 
[2K[33m[00152228ms][WARN][CELL]: URC already added with prefix: +CGEV: NW PDN D[0m
23:20:42.665 -> 
[2K[33m[00152228ms][WARN][CELL]: URC already added with prefix: +CGEV: ME PDN D[0m
23:20:42.665 -> 
[2K[33m[00152236ms][WARN][CELL]: URC already added with prefix: +QIURC: "pdpde[0m
23:20:42.665 -> 
[2K[39m[00152244ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:20:42.665 -> 
[2K[39m[00152249ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:20:43.743 -> 
[2K[39m[00153345ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:20:43.743 -> 
[2K[39m[00153345ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:20:44.868 -> 
[2K[39m[00154446ms][INFO][CELL]: AT TX ( 4): ATE0[0m
23:20:44.868 -> 
[2K[39m[00154446ms][INFO][CELL]: AT TX ( 1): <cr>[0m
23:20:45.946 -> 
[2K[39m[00155547ms][INFO][CELL]: Continue after 128 seconds[0m

paul-szczepanek-arm, 2021-03-10:
Doesn't look like your modem ever powers on.

Are you sure you're using the latest cellular example? Please clone this repo, `git checkout development`, `mbed deploy`. Change the mbed_app.json options for tracing to true, set trace level to debug, enable at traces. Compile and flash. Connect serial and dump it into a file, attach the file with the log to the issue together with your json.

Here's a normal log:

```
mbed-os-example-cellular
[00001000ms][INFO][CELL]: New CellularContext  (0x20007B58)
Establishing connection
[00001000ms][INFO][CELL]: CellularContext connect
[00001000ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00001000ms][INFO][CELL]: Start connecting (timeout 1000 ms)
[00001062ms][INFO][CELL]: AT TX ( 2): AT
[00001121ms][INFO][CELL]: AT TX ( 1): <cr>
[00002063ms][DBG ][CELL]: AT error -3012
[00002063ms][INFO][CELL]: AT TX ( 2): AT
[00002063ms][INFO][CELL]: AT TX ( 1): <cr>
[00003064ms][DBG ][CELL]: AT error -3012
[00003064ms][DBG ][CELL]: Init => Power
[00003064ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00003064ms][INFO][CELL]: Modem power ON (timeout 1000 ms)
[00004068ms][DBG ][CELL]: Power => Device ready
[00004068ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00004068ms][DBG ][CELL]: Device was not ready, calling soft_power_on()
[00005368ms][DBG ][CELL]: AT flush
[00005368ms][INFO][CELL]: AT TX ( 4): ATE0
[00005368ms][INFO][CELL]: AT TX ( 1): <cr>
[00006369ms][DBG ][CELL]: AT error -3012
[00006369ms][DBG ][CELL]: Wait 100ms to init modem
[00006469ms][DBG ][CELL]: AT flush
[00006469ms][INFO][CELL]: AT TX ( 4): ATE0
[00006469ms][INFO][CELL]: AT TX ( 1): <cr>
[00007470ms][DBG ][CELL]: AT error -3012
[00007470ms][DBG ][CELL]: Wait 100ms to init modem
[00007570ms][DBG ][CELL]: AT flush
[00007570ms][INFO][CELL]: AT TX ( 4): ATE0
[00007570ms][INFO][CELL]: AT TX ( 1): <cr>
[00008571ms][DBG ][CELL]: AT error -3012
[00008571ms][DBG ][CELL]: Wait 100ms to init modem
[00008671ms][DBG ][CELL]: Device ready: retry 0/10
[00008671ms][DBG ][CELL]: callback: CellularStateRetryEvent, err: 0, data: 4096, retrycount: 0
[00008671ms][INFO][CELL]: Continue after 1 seconds
[00009682ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00009682ms][DBG ][CELL]: Device was not ready, calling soft_power_on()
[00010982ms][WARN][CELL]: URC already added with prefix: +CGEV: NW DEACT
[00010982ms][WARN][CELL]: URC already added with prefix: +CGEV: ME DEACT
[00010982ms][WARN][CELL]: URC already added with prefix: +CGEV: NW PDN D
[00010989ms][WARN][CELL]: URC already added with prefix: +CGEV: ME PDN D
[00011081ms][DBG ][CELL]: AT flush
[00011133ms][INFO][CELL]: AT TX ( 4): ATE0
[00011193ms][INFO][CELL]: AT TX ( 1): <cr>
[00011254ms][INFO][CELL]: AT RX (11): atE0<cr><cr><ln>OK<cr><ln>
[00011337ms][INFO][CELL]: AT TX ( 9): AT+CMEE=1
[00011403ms][INFO][CELL]: AT TX ( 1): <cr>
[00011463ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[00011538ms][INFO][CELL]: AT TX ( 9): AT+CFUN=1
[00011604ms][INFO][CELL]: AT TX ( 1): <cr>
[00011664ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[00011739ms][INFO][CELL]: AT TX ( 7): AT+CGMI
[00011803ms][INFO][CELL]: AT TX ( 1): <cr>
[00011863ms][INFO][CELL]: AT RX (32): <cr><ln>+QIND: PB DONE<cr><ln><cr><ln>Quectel<cr><ln><cr><ln>O
[00011984ms][INFO][CELL]: AT RX ( 3): K<cr><ln>
[00012049ms][INFO][CELL]: Modem manufacturer: +QIND: PB DONE

Quectel

[00012144ms][INFO][CELL]: AT TX ( 7): AT+CGMM
[00012208ms][INFO][CELL]: AT TX ( 1): <cr>
[00012268ms][INFO][CELL]: AT RX (14): <cr><ln>UG96<cr><ln><cr><ln>OK<cr><ln>
[... 113 lines cut ...]
[00022912ms][INFO][CELL]: AT TX ( 1): <cr>
[00023865ms][WARN][CELL]: AT timeout
[00023865ms][DBG ][CELL]: AT error -3012
[00023865ms][DBG ][CELL]: consume_to_tag not found
[00023865ms][DBG ][CELL]: AT stop tag not found
[00023865ms][DBG ][CELL]: AT error -3012
[00023902ms][WARN][CELL]: AT timeout
[00023956ms][DBG ][CELL]: AT error -3012
[00024015ms][DBG ][CELL]: consume_to_tag not found
[00024083ms][DBG ][CELL]: AT stop tag not found
[00024149ms][DBG ][CELL]: AT error -3012
[00024207ms][INFO][CELL]: AT TX ( 8): AT+CMEE?
[00024272ms][INFO][CELL]: AT TX ( 1): <cr>
[00024332ms][INFO][CELL]: AT RX (32): <cr><ln>NO CARRIER<cr><ln><cr><ln>+CGEV: ME DEACT 
[00024441ms][INFO][CELL]: AT RX (32):   "IP","10.148.63.160",1<cr><ln><cr><ln>+CME
[00024543ms][INFO][CELL]: AT RX (12): E: 1<cr><ln><cr><ln>OK<cr><ln>
[00024630ms][DBG ][CELL]: callback: 0, ptr: 2
[00024694ms][INFO][CELL]: cellular_callback: PPP mode and NSAPI_STATUS_DISCONNECTED
[00024797ms][DBG ][CELL]: AT sync
[00024848ms][INFO][CELL]: AT TX ( 8): AT+CMEE?
[00024913ms][INFO][CELL]: AT TX ( 1): <cr>
[00024973ms][INFO][CELL]: AT RX (18): <cr><ln>+CMEE: 1<cr><ln><cr><ln>OK<cr><ln>
[00025073ms][INFO][CELL]: CellularContext disconnected
Success. Exiting
```

paul-szczepanek-arm, 2021-03-10:
I imagine the reason the modem doesn't power on is because there is no communication across the AT protocol. Check the correct pins are configured in the json.

adityathakekar, 2021-03-10:
Attaching the logs with json file .. 
using Dragion NB-iot shield quectel BG96

{
    "config": {
        "sock-type": "UDP",
        "echo-server-hostname": {
            "help": "Echo server host name.",
            "value": "\"18.197.48.88\""
        },
        "echo-server-port": {
            "help": "Echo server port number.",
            "value": 5681
        },
        "echo-device-port":
        {
            "help" : "Echo device port number.",
            "value" : 5685
        },
        "echo-connect-id":
        {
              "help": "Echo connect id",
              "value": 3
        },
        "trace-level": {
            "help": "Options are TRACE_LEVEL_ERROR,TRACE_LEVEL_WARN,TRACE_LEVEL_INFO,TRACE_LEVEL_DEBUG",
            "macro_name": "MBED_TRACE_MAX_LEVEL",
            "value": "TRACE_LEVEL_INFO"
        }
  },
    "target_overrides": {
        "*": {
            "target.network-default-interface-type": "CELLULAR",
            "mbed-trace.enable": true,
            "lwip.ipv4-enabled": true,
            "ppp.ipv4-enabled": true,
            "lwip.ipv6-enabled": true,
            "ppp.ipv6-enabled": true,
            "lwip.ethernet-enabled": false,
            "lwip.ppp-enabled": true,
            "lwip.tcp-enabled": true,
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 9600,
            "platform.default-serial-baud-rate": 9600,
            "platform.stdio-buffered-serial": true,
            "cellular.debug-at": true,
            "cellular.use-apn-lookup": true,
            "nsapi.default-cellular-sim-pin": null,
            "nsapi.default-cellular-plmn": null,
            "nsapi.default-cellular-apn": "\"iot.1nce.net\"",
            "nsapi.default-cellular-username": null,
            "nsapi.default-cellular-password": null,
            "QUECTEL_BG96.tx": "D1",
            "QUECTEL_BG96.rx": "D0",
            "QUECTEL_BG96.provide-default": true

        },
        "DISCO_L496AG": {
            "target.macros_add": [
                "CELLULAR_DEVICE=STModCellular"
            ],
            "target.components_add": ["STMOD_CELLULAR"],
            "stmod_cellular.provide-default": "true"
        }
    }
}

[stm32-logs.txt](https://github.com/ARMmbed/mbed-os-example-cellular/files/6116170/stm32-logs.txt)

paul-szczepanek-arm, 2021-03-10:
OK, looks like you don't have communicatin between the board and the shield. I think you need to configure the QUECTEL_BG96. Look in https://github.com/ARMmbed/mbed-os/blob/master/connectivity/drivers/cellular/QUECTEL/BG96/mbed_lib.json for values you need to provide.

This doesn't look like an issue with the example though. I recommend the forums or if there's something wrong with BG96 the mbed-os repo.

adityathakekar, 2021-03-10:
@paul-szczepanek-arm thanks for your time. closing this issue.

closed 2021-03-10 by adityathakekar