ARMmbed/mbed-os-example-cellular #222: u-blox Nina b112 error + Sara G350

tcpipchip, 2021-06-14:
Sir, 
How are you ?
After config

        "nsapi.default-cellular-apn": "\"java.com.br.br\"",
            "nsapi.default-cellular-username":  "\"claro\"",
            "nsapi.default-cellular-password":  "\"claro\"",
            "GENERIC_AT3GPP.tx": "p8",
	    "GENERIC_AT3GPP.rx": "p11",
	"GENERIC_AT3GPP.baudrate":115200
        },
        "NRF52_DK": {
            "target.macros_add": ["UBX_MDM_SARA_G3XX", "UBX_MDM_SARA_G350"]
        }

I am getting the error 

mbed-os-example-cellular
Failed to get_default_instance()

Can you help me ?

0xc0170, 2021-06-14:
The error comes from the example

```
#if MBED_CONF_APP_SOCK_TYPE == NONIP
    NetworkInterface *net = CellularContext::get_default_nonip_instance();
#else
    NetworkInterface *net = CellularContext::get_default_instance();
#endif
```

You should check what is being returned inside this default instance function (I assume you get into ip one).

AnttiKauppila, 2021-06-14:
Have you configured the modem to provide the default instance?
You need to override and set one of the following to true in your mbed_app.json: 
https://github.com/ARMmbed/mbed-os/blob/master/connectivity/drivers/cellular/UBLOX/PPP/mbed_lib.json#L24-L26 for PPP mode or 
https://github.com/ARMmbed/mbed-os/blob/master/connectivity/drivers/cellular/UBLOX/AT/mbed_lib.json#L24-L26 for AT mode

tcpipchip, 2021-06-14:
Sir
Thank you for the fast help! I hope soon test in 4 u-blox modem models that i have here! I intend to make blogs about then using the PPP.
Well, getting progress here!

tcpipchip, 2021-06-14:
[image]
Look

tcpipchip, 2021-06-14:
getting more progresss!
But i need some TIP
[image]

tcpipchip, 2021-06-14:
Here a link
https://youtu.be/FVG1x1xtzto

tcpipchip, 2021-06-15:
Hi, i tried too other servers...got connection, but cant send and receive data

some suggestion team ?

tcpipchip, 2021-06-17:
Sir
Can you see if am i doing something wrong ?
`/*
 * Copyright (c) 2017 ARM Limited. All rights reserved.
 * SPDX-License-Identifier: Apache-2.0
 * Licensed under the Apache License, Version 2.0 (the License); you may
 * not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 * http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an AS IS BASIS, WITHOUT
 * WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#include "mbed.h"
#include "CellularNonIPSocket.h"
#include "UDPSocket.h"
#include "TCPSocket.h"
#include "cellular_demo_tracing.h"

/* configuration choices in mbed_app.json */
#define UDP 0
#define TCP 1
#define NONIP 2

#if MBED_CONF_APP_SOCK_TYPE == TCP
static constexpr char SOCKET_TYPE[] = "TCP";
#elif MBED_CONF_APP_SOCK_TYPE == UDP
static constexpr char SOCKET_TYPE[] = "UDP";
#elif MBED_CONF_APP_SOCK_TYPE == NONIP
static constexpr char SOCKET_TYPE[] = "CellularNonIP";
#endif
static const char ECHO_HOSTNAME[] = MBED_CONF_APP_ECHO_SERVER_HOSTNAME;

class CellularDemo {
    static constexpr uint8_t RETRY_COUNT = 3;

public:
    CellularDemo(NetworkInterface &network)
        : _net(network)
    { }

    ~CellularDemo() { }

    /** Run the cellular demo. */
    void run()
    {
        /* sim pin, apn, credentials and possible plmn are taken automatically from json
         * when using NetworkInterface::set_default_parameters() */
        _net.set_default_parameters();

        nsapi_size_or_error_t ret = NSAPI_ERROR_NO_CONNECTION;

        if (connect_cellular()) {
            /* ping echo server */
            if (!test_send_and_receive()) {
                printf("Sending and received data failed.\n");
                wait_us(1000000);
            }

            ret = _net.disconnect();

            if (ret != NSAPI_ERROR_OK) {
                printf("Disconnect failed (error: %d).\n", ret);
                wait_us(1000000);
            }
[... 180 lines cut ...]

#if MBED_CONF_APP_SOCK_TYPE == NONIP
    NetworkInterface *net = CellularContext::get_default_nonip_instance();
#else
    NetworkInterface *net = CellularContext::get_default_instance();
#endif

    if (net) {
        CellularDemo example(*net);
        example.run();
    } else {
        printf("Failed to get_default_instance()\n");
        wait_us(1000000);
    }

    trace_close();

    return 0;
}
`

Getting always this error
[image]

Tested in 4 modems

paul-szczepanek-arm, 2021-06-18:
Enabling tracing might help you diagnose the problem.

tcpipchip, 2021-06-19:
hi @paul-szczepanek-arm 

Strange that there that PPP CONNECTION is estabilshed, the DNS resolver work, make the SOCKET CONNECTION, but not send and receive.
To me,  looks some UART problem, but i dont know what...
Something weird, because in the main.cpp, i had to include after each printf, and  wait_us(1000000); otherwise, overrun on my uart receive
printf("Disconnect failed (error: %d).\n", ret);
wait_us(1000000);

Btw something is happeing in the MODEM UART too

If i enable the trace, i have the UART OVERRUN and i can see anything :(

Now i am testing on SARA G450 connected to a NINA B112 (NRF52832)

Do the PPP is using handshacking ?

Thanks!

tcpipchip, 2021-07-03:
Must i try to change to PAP or CHAP to works ?
Did you try already with NRF52 ?

tcpipchip, 2021-07-06:
Today i did try the AT mode
[image]
same problem
looks that the server is no ECHOing the my TEST data too :(
very very very weird

tcpipchip, 2021-07-06:
I found something very interesting!
If you put in 

UBLOX_AT__celularStack.cpp the following instruction

	 printf("%s\n","EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE");

UBLOX_AT_CellularStack::UBLOX_AT_CellularStack(ATHandler &atHandler, int cid, nsapi_ip_stack_t stack_type, AT_CellularDevice &device) :
    AT_CellularStack(atHandler, cid, stack_type, device)
{
    // URC handlers for sockets
	
	 printf("%s\n","EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE");
	
    _at.set_urc_handler("+UUSORD:", callback(this, &UBLOX_AT_CellularStack::UUSORD_URC));
    _at.set_urc_handler("+UUSORF:", callback(this, &UBLOX_AT_CellularStack::UUSORF_URC));
    _at.set_urc_handler("+UUSOCL:", callback(this, &UBLOX_AT_CellularStack::UUSOCL_URC));
    _at.set_urc_handler("+UUPSDD:", callback(this, &UBLOX_AT_CellularStack::UUPSDD_URC));
}

But....all AT commands sent to MODEM work perfectly!

Can you give some tip about ?
[image]